micropython/micropython #3603: mphal.h - #ifndef used to check for existence of a function?

adritium, 2018-02-06:
It appears that `#ifndef` is used to check for existence of a function; [that is not allowed in C](https://stackoverflow.com/questions/6234755/does-ifdef-or-other-preprocessor-directives-work-for-function-declarations-t).
Only existence of MACROS can be checked with `#ifndef`.

```
#ifndef mp_hal_stdout_tx_str
void mp_hal_stdout_tx_str(const char *str);
#endif
```

If `mp_hal_stdout_tx_str` - and all the other arguments of `#ifndef` - are meant to be macros, I suggest the entries in this file be changed to indicate they are macros i.e. by CAPITALZING them.

dpgeorge, 2018-02-07:
> If mp_hal_stdout_tx_str - and all the other arguments of #ifndef in this file - are meant to be macros,

They are either functions or macros, depending on how the port wants to configure it.  The idea is that you either 1) implement the function with the same signature; 2) define the macro of the same name to provide whatever implementation you need (eg some inline code).  See eg https://github.com/micropython/micropython/blob/master/ports/cc3200/hal/cc3200_hal.h#L67-L68 for an example of the latter use.

> I suggest the entries in this file be changed to indicate they are macros i.e. by CAPITALZING them.

They are for most ports functions, so that's why they are lower case.

> Why not follow the same pattern as in mpconfig.h?
> If macro doesn't exist, create its definition here:

But the definition is not known in mphal.h.  It's only known by a port.

dpgeorge, 2018-03-07:
I assume this point was clarified.

closed 2018-03-07 by dpgeorge