micropython/micropython #3604: TouchPad GPIO pins swapped

rdagger, 2018-02-08:
Looks like GPIO 32 and 33 are swapped with respect to the TouchPads:
```
t32 = TouchPad(Pin(32))
t33 = TouchPad(Pin(33))
```
t32 in above code responds to touches on GPIO33 and t33 responds to GPIO32.
However, the machine pin I/O is correct:
```
p32 = Pin(32, Pin.IN, Pin.PULL_UP)
p33 = Pin(33, Pin.IN, Pin.PULL_UP)
```
p32 responds to input on GPIO32 and p33 responds to GPIO33 (based on the GPIO markings on the Lolin32 board.)

dpgeorge, 2018-02-18:
Thanks for the report.  I can confirm the issue (also on a Lolin32 board).  Should now be fixed by 5a82ba8e073f847985595a46fb2f0c12f4389bbc

rdagger, 2018-02-23:
On a related note, it doesn't seem like TouchPad.config() method does anything.  I performed a test with config values between 0 and 1000 and the read results don't have any discernible differences:
```
Threshold0: 1144
sensitivity 0
Touch0: 1062, Diff: 82, Ratio: 92.83217%.
sensitivity increased to 100
Touch0: 1076, Diff: 68, Ratio: 94.05595%.
sensitivity increased to 200
Touch0: 1044, Diff: 100, Ratio: 91.25875%.
sensitivity increased to 300
Touch0: 1067, Diff: 77, Ratio: 93.26922%.
sensitivity increased to 400
Touch0: 1070, Diff: 74, Ratio: 93.53148%.
sensitivity increased to 500
Touch0: 1053, Diff: 91, Ratio: 92.04546%.
sensitivity increased to 600
Touch0: 1077, Diff: 67, Ratio: 94.14336%.
sensitivity increased to 700
Touch0: 1061, Diff: 83, Ratio: 92.74476%.
sensitivity increased to 800
Touch0: 1048, Diff: 96, Ratio: 91.60839%.
sensitivity increased to 900
Touch0: 1072, Diff: 72, Ratio: 93.70629%.
sensitivity increased to 1000
```

Here's my test code:
```
"""TouchPad Config Test."""
from time import sleep
from machine import Pin, TouchPad

touch0 = TouchPad(Pin(4))
threshold0 = []

# Scan TouchPad 12 times for calibration
for x in range(12):
    threshold0.append(touch0.read())
    sleep(.1)

# Store average threshold value
threshold0 = sum(threshold0) // len(threshold0)
print('Threshold0: {0}'.format(threshold0))

sensitivity = 0
touch0.config(sensitivity)
print('sensitivity {0}'.format(sensitivity))

try:
    while True:
        capacitance0 = touch0.read()
        cap_ratio0 = capacitance0 / threshold0
        # Check if a Touchpad is pressed
        if .40 < cap_ratio0 < .95:
            print('Touch0: {0}, Diff: {1}, Ratio: {2}%.'.format(
                  capacitance0, threshold0 - capacitance0, cap_ratio0 * 100))
            sleep(.25)  # Debounce press
            sensitivity += 100
            touch0.config(sensitivity)
            print('sensitivity increased to {0}'.format(sensitivity))

except KeyboardInterrupt:
    print('\nCtrl-C pressed.  Exiting...')

```

dpgeorge, 2018-03-05:
> I just tried today's build [...] and the same problem persists.

I can confirm that the problem is still there... strange indeed.

dpgeorge, 2018-03-05:
It looks like there is an issue with the hardware, that touch pad 8 and 9 have some registers swapped between them.  And this is now fixed in the ESP IDF, so there's no need to swap pins 32 and 33 in the code here.  The patch was reverted in d4470af239ffb536f3fbf610060ddb62789d3045 and it now seems to work properly.

dpgeorge, 2018-03-05:
> it doesn't seem like TouchPad.config() method does anything

I think config() sets the threshold for an interrupt on a touch pin.  Interrupts are not yet supported on touch pins so indeed config() won't change anything from the point of view of the read() function.

closed 2018-03-05 by dpgeorge