micropython/micropython #3605: make axtls fails for ESP8266

larsrinn, 2018-02-10:
Hi,

I'm trying to build from sources for the ESP8266. Building for unix worked fine, but I can't figure what I'm doing wrong, so I came to the conclusion that it must be a bug or something missing in the documentation...

So what did I do:
* `git clone` the repo and `cd` into it
* `git submodule update --init`
* `cd` to `ports/esp8266`
* `sudo make axtls` (tried it without sudo as well)
* This yields the errror:
```
make[1]: Leaving directory '/mnt/c/Users/lmrin/2/micropython/lib/axtls'
Makefile:202: recipe for target 'build/libaxtls.a' failed
make: *** [build/libaxtls.a] Error 2
```

I tried this both on a Ubuntu 16.04 system inside WSL and on Ubuntu Mate on a Raspberry Pi. I made sure, `axtls` was not build for another architecture already by running `make clean` inside `libs/axtls`

pmp-p, 2018-02-10:
did you try with "make V=1" to get more info ?

dpgeorge, 2018-02-11:
@larsrinn to help debug the issue you'll need to provide more detail of the output, not just the last lines.

lucianreynolds, 2018-02-11:
I am getting the same error as @larsrinn. Here is the output with make V=1:

`vagrant@vagrant-ubuntu-trusty-64:~/micropython/ports/esp8266$ sudo make V=1 axtls
cd ../../lib/axtls; cp config/upyconfig config/.config
cd ../../lib/axtls; make oldconfig -B
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C config/scripts/config conf
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/config/scripts/config'
cp ../../scripts/config/zconf.tab.h_shipped zconf.tab.h
gcc -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -I. -c ../../scripts/config/conf.c -o conf.o
../../scripts/config/conf.c: In function ‘conf_string’:
../../scripts/config/conf.c:164:20: warning: variable ‘help’ set but not used [-Wunused-but-set-variable]
  const char *def, *help;
                    ^
../../scripts/config/conf.c: In function ‘conf_sym’:
../../scripts/config/conf.c:198:6: warning: variable ‘type’ set but not used [-Wunused-but-set-variable]
  int type;
      ^
../../scripts/config/conf.c: In function ‘conf_choice’:
../../scripts/config/conf.c:273:6: warning: variable ‘type’ set but not used [-Wunused-but-set-variable]
  int type;
      ^
../../scripts/config/conf.c: In function ‘conf_askvalue’:
../../scripts/config/conf.c:94:8: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
   fgets(line, 128, stdin);
        ^
../../scripts/config/conf.c: In function ‘conf_choice’:
../../scripts/config/conf.c:350:9: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
    fgets(line, 128, stdin);
         ^
cp ../../scripts/config/zconf.tab.c_shipped zconf.tab.c
cp ../../scripts/config/lex.zconf.c_shipped lex.zconf.c
gcc -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -I../../scripts/config -I. -c zconf.tab.c -o zconf.tab.o
In file included from zconf.tab.c:2123:0:
lex.zconf.c:2969:16: warning: ‘input’ defined but not used [-Wunused-function]
     static int input  (void)
                ^
gcc  conf.o zconf.tab.o -o conf
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/config/scripts/config'
#
# using defaults found in config/.config
#
config/.config:46: trying to assign nonexistent symbol CONFIG_SSL_ENABLE_V23_HANDSHAKE
*
* axTLS Configuration
*
Platform
> 1. Linux (CONFIG_PLATFORM_LINUX)
  2. Cygwin (CONFIG_PLATFORM_CYGWIN)
  3. Win32 (CONFIG_PLATFORM_WIN32)
choice[1-3]: 1
*
* General Configuration
*
axTLS installation prefix (PREFIX) [/usr/local] /usr/local
Build axTLS with Debugging symbols (CONFIG_DEBUG) [N/y/?] n
  Strip unwanted sections from elf binaries (CONFIG_STRIP_UNWANTED_SECTIONS) [N/y/?] n
Any extra CFLAGS options for the compiler? (CONFIG_EXTRA_CFLAGS_OPTIONS) [] 
Any extra LDFLAGS options for the compiler? (CONFIG_EXTRA_LDFLAGS_OPTIONS) [] 
*
* SSL Library
*
Mode
  1. Server only - no verification (CONFIG_SSL_SERVER_ONLY)
  2. Server only - with verification (CONFIG_SSL_CERT_VERIFICATION)
  3. Client/Server enabled with diagnostics (CONFIG_SSL_FULL_MODE)
> 4. Skeleton mode - the smallest server mode (CONFIG_SSL_SKELETON_MODE)
choice[1-4]: 4
Server enabled (CONFIG_SSL_ENABLE_SERVER) [Y/n/?] y
Client enabled (CONFIG_SSL_ENABLE_CLIENT) [Y/n/?] y
[... 64 lines cut ...]
make -C vbnet clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/vbnet'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/vbnet'
make -C java clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/java'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/java'
make -C perl clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/perl'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/perl'
make -C lua clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/lua'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/lua'
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings'
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
cd ../../lib/axtls; make all CC="xtensa-lx106-elf-gcc" LD="xtensa-lx106-elf-ld" AR="xtensa-lx106-elf-ar" CFLAGS_EXTRA="-fsingle-precision-constant -Wdouble-promotion -D__ets__ -DICACHE_FLASH -fno-inline-functions -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32 -DLWIP_OPEN_SRC -Dabort=abort_ -DRT_MAX_PLAIN_LENGTH=1024 -DRT_EXTRA=4096"
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C crypto
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/crypto'
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
make[2]: *** No rule to make target `.depend', needed by `all'.  Stop.
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/crypto'
make[1]: *** [target] Error 2
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
make: *** [build/libaxtls.a] Error 2
`

hthuong09, 2018-02-12:
@lucianreynolds please put it into a code block.

lucianreynolds, 2018-02-12:
Yikes. I did forward ticks. My mistake.
```bash
vagrant@vagrant-ubuntu-trusty-64:~/micropython/ports/esp8266$ sudo make V=1 axtls
cd ../../lib/axtls; cp config/upyconfig config/.config
cd ../../lib/axtls; make oldconfig -B
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C config/scripts/config conf
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/config/scripts/config'
cp ../../scripts/config/zconf.tab.h_shipped zconf.tab.h
gcc -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -I. -c ../../scripts/config/conf.c -o conf.o
../../scripts/config/conf.c: In function ‘conf_string’:
../../scripts/config/conf.c:164:20: warning: variable ‘help’ set but not used [-Wunused-but-set-variable]
  const char *def, *help;
                    ^
../../scripts/config/conf.c: In function ‘conf_sym’:
../../scripts/config/conf.c:198:6: warning: variable ‘type’ set but not used [-Wunused-but-set-variable]
  int type;
      ^
../../scripts/config/conf.c: In function ‘conf_choice’:
../../scripts/config/conf.c:273:6: warning: variable ‘type’ set but not used [-Wunused-but-set-variable]
  int type;
      ^
../../scripts/config/conf.c: In function ‘conf_askvalue’:
../../scripts/config/conf.c:94:8: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
   fgets(line, 128, stdin);
        ^
../../scripts/config/conf.c: In function ‘conf_choice’:
../../scripts/config/conf.c:350:9: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
    fgets(line, 128, stdin);
         ^
cp ../../scripts/config/zconf.tab.c_shipped zconf.tab.c
cp ../../scripts/config/lex.zconf.c_shipped lex.zconf.c
gcc -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -I../../scripts/config -I. -c zconf.tab.c -o zconf.tab.o
In file included from zconf.tab.c:2123:0:
lex.zconf.c:2969:16: warning: ‘input’ defined but not used [-Wunused-function]
     static int input  (void)
                ^
gcc  conf.o zconf.tab.o -o conf
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/config/scripts/config'
#
# using defaults found in config/.config
#
config/.config:46: trying to assign nonexistent symbol CONFIG_SSL_ENABLE_V23_HANDSHAKE
*
* axTLS Configuration
*
Platform
> 1. Linux (CONFIG_PLATFORM_LINUX)
  2. Cygwin (CONFIG_PLATFORM_CYGWIN)
  3. Win32 (CONFIG_PLATFORM_WIN32)
choice[1-3]: 1
*
* General Configuration
*
axTLS installation prefix (PREFIX) [/usr/local] /usr/local
Build axTLS with Debugging symbols (CONFIG_DEBUG) [N/y/?] n
  Strip unwanted sections from elf binaries (CONFIG_STRIP_UNWANTED_SECTIONS) [N/y/?] n
Any extra CFLAGS options for the compiler? (CONFIG_EXTRA_CFLAGS_OPTIONS) [] 
Any extra LDFLAGS options for the compiler? (CONFIG_EXTRA_LDFLAGS_OPTIONS) [] 
*
* SSL Library
*
Mode
  1. Server only - no verification (CONFIG_SSL_SERVER_ONLY)
  2. Server only - with verification (CONFIG_SSL_CERT_VERIFICATION)
  3. Client/Server enabled with diagnostics (CONFIG_SSL_FULL_MODE)
> 4. Skeleton mode - the smallest server mode (CONFIG_SSL_SKELETON_MODE)
choice[1-4]: 4
Server enabled (CONFIG_SSL_ENABLE_SERVER) [Y/n/?] y
Client enabled (CONFIG_SSL_ENABLE_CLIENT) [Y/n/?] y
[... 64 lines cut ...]
make -C vbnet clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/vbnet'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/vbnet'
make -C java clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/java'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/java'
make -C perl clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/perl'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/perl'
make -C lua clean
make[3]: Entering directory `/home/vagrant/micropython/lib/axtls/bindings/lua'
make[3]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings/lua'
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/bindings'
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
cd ../../lib/axtls; make all CC="xtensa-lx106-elf-gcc" LD="xtensa-lx106-elf-ld" AR="xtensa-lx106-elf-ar" CFLAGS_EXTRA="-fsingle-precision-constant -Wdouble-promotion -D__ets__ -DICACHE_FLASH -fno-inline-functions -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32 -DLWIP_OPEN_SRC -Dabort=abort_ -DRT_MAX_PLAIN_LENGTH=1024 -DRT_EXTRA=4096"
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C crypto
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/crypto'
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
make[2]: *** No rule to make target `.depend', needed by `all'.  Stop.
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/crypto'
make[1]: *** [target] Error 2
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
make: *** [build/libaxtls.a] Error 2
```

dpgeorge, 2018-02-12:
> /bin/sh: 1: xtensa-lx106-elf-gcc: not found

@lucianreynolds It looks like you need to install the ESP8266 toolchain, found at https://github.com/pfalcon/esp-open-sdk

lucianreynolds, 2018-02-14:
I'm pretty sure that I compiled that in the earlier step:

```
vagrant@vagrant-ubuntu-trusty-64:~$ cd esp-open-sdk/
vagrant@vagrant-ubuntu-trusty-64:~/esp-open-sdk$ make STANDALONE=y
Makefile:184: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:179: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory `/home/vagrant/esp-open-sdk/crosstool-NG'
../Makefile:184: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:179: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... objcopy
checking for absolute path to objcopy... /usr/bin/objcopy
checking for gobjdump... no
checking for objdump... objdump
checking for absolute path to objdump... /usr/bin/objdump
checking for greadelf... no
checking for readelf... readelf
checking for absolute path to readelf... /usr/bin/readelf
checking for gperf... gperf
checking for absolute path to gperf... /usr/bin/gperf
checking for bison... bison
checking for flex... flex
checking for makeinfo... makeinfo
checking for cut... cut
checking for stat... stat
checking for readlink... readlink
checking for wget... wget
checking for tar... tar
checking for gzip... gzip
checking for bzip2... bzip2
checking for help2man... help2man
checking for gpatch... no
checking for patch... patch
checking for absolute path to patch... /usr/bin/patch
checking for bash >= 3.1... /bin/bash
checking for GNU awk... /usr/bin/gawk
checking for GNU make >= 3.80... /usr/bin/make
checking whether /usr/bin/make sets $(MAKE)... yes
checking for make 3.81... yes
checking for GNU libtool >= 1.5.26... /usr/bin/libtool
checking for GNU libtoolize >= 1.5.26... /usr/bin/libtoolize
checking for GNU automake >= 1.10... /usr/bin/automake
checking for the type used in gperf declarations... unsigned int
checking for xz... xz
[... 1049 lines cut ...]
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/udp.o lwip/core/udp.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/autoip.o lwip/core/ipv4/autoip.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/icmp.o lwip/core/ipv4/icmp.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/igmp.o lwip/core/ipv4/igmp.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/inet.o lwip/core/ipv4/inet.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/inet_chksum.o lwip/core/ipv4/inet_chksum.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/ip_addr.o lwip/core/ipv4/ip_addr.c
/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -DMEMCPY=memcpy -DSMEMCPY=memcpy -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32    -c -o lwip/core/ipv4/ip.o lwip/core/ipv4/ip.c
[... cut ...]
cp liblwip_open.a /home/vagrant/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/
make[1]: Leaving directory `/home/vagrant/esp-open-sdk/esp-open-lwip'
cp -a esp-open-lwip/include/arch esp-open-lwip/include/lwip esp-open-lwip/include/netif \
	    esp-open-lwip/include/lwipopts.h \
	    /home/vagrant/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/

Xtensa toolchain is built, to use it:

export PATH=/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin:$PATH

Espressif ESP8266 SDK is installed, its libraries and headers are merged with the toolchain
```

dpgeorge, 2018-02-14:
@lucianreynolds did you add the location of the toolchain binaries to your path (see the `export PATH=...` line at the end of the output above)?  If so then you should be able to execute `xtensa-lx106-elf-gcc` on the command line.

lucianreynolds, 2018-02-15:
Before I go any further, I want to thank you for working through this with me.

After compiling I added the toolchain to my vagrant profile:

```
echo "PATH=$(pwd)/xtensa-lx106-elf/bin:\$PATH" >> ~/.profile
```

I exited and restarted vagrant and checked to see if the toolchain was added:

```vagrant@vagrant-ubuntu-trusty-64:~/micropython/ports/esp8266$ echo $PATH
/home/vagrant/xtensa-lx106-elf/bin:/home/vagrant/micropython/ports/esp8266/xtensa-lx106-elf/bin:/home/vagrant/xtensa-esp32-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
```

Next I installed the dependencies:

```
vagrant@vagrant-ubuntu-trusty-64:~$ cd ~/micropython/
vagrant@vagrant-ubuntu-trusty-64:~/micropython$ git submodule update --init
Submodule 'lib/axtls' (https://github.com/pfalcon/axtls) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lwip' (https://git.savannah.gnu.org/r/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Cloning into 'lib/axtls'...
remote: Counting objects: 3179, done.
remote: Total 3179 (delta 0), reused 0 (delta 0), pack-reused 3179
Receiving objects: 100% (3179/3179), 3.81 MiB | 7.11 MiB/s, done.
Resolving deltas: 100% (2333/2333), done.
Checking connectivity... done.
Submodule path 'lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
Cloning into 'lib/berkeley-db-1.xx'...
remote: Counting objects: 270, done.
Receiving objects: 100% (270/270), 332.23 KiB | 0 bytes/s, done.
remote: Total 270 (delta 0), reused 0 (delta 0), pack-reused 270
Resolving deltas: 100% (127/127), done.
Checking connectivity... done.
Submodule path 'lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
Cloning into 'lib/libffi'...
remote: Counting objects: 11199, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 11199 (delta 0), reused 2 (delta 0), pack-reused 11194
Receiving objects: 100% (11199/11199), 6.96 MiB | 6.83 MiB/s, done.
Resolving deltas: 100% (6956/6956), done.
Checking connectivity... done.
Submodule path 'lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Cloning into 'lib/lwip'...
remote: Counting objects: 47934, done.
remote: Compressing objects: 100% (11341/11341), done.
remote: Total 47934 (delta 36107), reused 47796 (delta 36019)
Receiving objects: 100% (47934/47934), 8.99 MiB | 7.52 MiB/s, done.
Resolving deltas: 100% (36107/36107), done.
Checking connectivity... done.
Submodule path 'lib/lwip': checked out '5b8b5d459e7dd890724515bbfad86c705234f9ec'
Cloning into 'lib/stm32lib'...
remote: Counting objects: 1746, done.
remote: Total 1746 (delta 0), reused 0 (delta 0), pack-reused 1746
Receiving objects: 100% (1746/1746), 6.01 MiB | 6.49 MiB/s, done.
Resolving deltas: 100% (1416/1416), done.
Checking connectivity... done.
Submodule path 'lib/stm32lib': checked out 'd2bcfda543d3b99361e44112aca929225bdcc07f'
vagrant@vagrant-ubuntu-trusty-64:~/micropython$ make -C mpy-cross
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Entering directory `/home/vagrant/micropython/mpy-cross'
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
[... 145 lines cut ...]
CC ../py/../extmod/uos_dupterm.c
CC ../py/../lib/embed/abort_.c
CC ../py/../lib/utils/printf.c
CC main.c
CC gccollect.c
LINK mpy-cross
   text	   data	    bss	    dec	    hex	filename
 131150	    808	    872	 132830	  206de	mpy-cross
make: Leaving directory `/home/vagrant/micropython/mpy-cross'
```

Lastly I tried to sudo make that bad boy and caught these errors that tanked the process:
```
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C crypto
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/crypto'
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
make[2]: *** No rule to make target `.depend', needed by `all'.  Stop.
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/crypto'
make[1]: *** [target] Error 2
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
make: *** [build/libaxtls.a] Error 2
vagrant@vagrant-ubuntu-trusty-64:~/micropython/ports/esp8266$
```
I am not experienced enough to catch where the error lies here....

cefn, 2018-02-15:
The command pwd is 'print working directory'. It outputs the directory you were in when you ran it. Therefore your PATH environment variable from your profile is now full of links to non-existent xtensa-lx106-elf/bin subdirectories since you seem to have been running the command from various places, not just the top folder of the esp-open-sdk repository.

The moment the PATH contains a link to the actual directory (wherever the xtensa-lx106-elf/bin directory actually exists on your machine) I think it is likely to build correctly. 

It is a subdirectory of the esp toolchain you built so most likely /home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin from the information you've shared.

I believe if you run the following line in a console

```
export PATH=/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/:$PATH
```

...then run... 

```
make clean; make axtls; make
```

...it will likely work.

For tidiness you will probably need to tidy up your .profile file from all the random folders which have now been permanently added to your path.

lucianreynolds, 2018-02-17:
I think that we are making progress here, even though I am still hitting the same wall. I followed your advice and now I think that the path is correct. Here is my .profile:

```
# ~/.profile: executed by the command interpreter for login shells.
# This file is not read by bash(1), if ~/.bash_profile or ~/.bash_login
# exists.
# see /usr/share/doc/bash/examples/startup-files for examples.
# the files are located in the bash-doc package.

# the default umask is set in /etc/profile; for setting the umask
# for ssh logins, install and configure the libpam-umask package.
#umask 022

# if running bash
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
        . "$HOME/.bashrc"
    fi
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi

PATH=/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/:$PATH
```
I even went into my Vagrantfile and doubled the amount of v.memory so I'm still at a loss as to why the make axtls portion of the process is failing. Does my profile above look right? 

I am still seeing these two errors at the end where everything goes south.
```
make[1]: Entering directory `/home/vagrant/micropython/lib/axtls'
make -C crypto
make[2]: Entering directory `/home/vagrant/micropython/lib/axtls/crypto'
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
make[2]: *** No rule to make target `.depend', needed by `all'.  Stop.
make[2]: Leaving directory `/home/vagrant/micropython/lib/axtls/crypto'
make[1]: *** [target] Error 2
make[1]: Leaving directory `/home/vagrant/micropython/lib/axtls'
make: *** [build/libaxtls.a] Error 2
vagrant@vagrant-ubuntu-trusty-64:~/micropython/ports/esp8266$ cd home/vagrant/micropython/lib/axtls/crypto'
```

cefn, 2018-02-17:
Is there a file called xtensa-lx106-elf-gcc in the folder called /home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/ ?

If not, then you need to go back a stage and make sure it was built when you followed the instructions to build the esp-open-sdk

lucianreynolds, 2018-02-17:
I got it. 
```
vagrant@vagrant-ubuntu-trusty-64:~/esp-open-sdk/xtensa-lx106-elf/bin$ ls
esptool.py                     xtensa-lx106-elf-gcc-ranlib
xtensa-lx106-elf-addr2line     xtensa-lx106-elf-gcov
xtensa-lx106-elf-ar            xtensa-lx106-elf-gdb
xtensa-lx106-elf-as            xtensa-lx106-elf-gprof
xtensa-lx106-elf-c++           xtensa-lx106-elf-ld
xtensa-lx106-elf-cc            xtensa-lx106-elf-ld.bfd
xtensa-lx106-elf-c++filt       xtensa-lx106-elf-nm
xtensa-lx106-elf-cpp           xtensa-lx106-elf-objcopy
xtensa-lx106-elf-ct-ng.config  xtensa-lx106-elf-objdump
xtensa-lx106-elf-elfedit       xtensa-lx106-elf-ranlib
xtensa-lx106-elf-g++           xtensa-lx106-elf-readelf
xtensa-lx106-elf-gcc           xtensa-lx106-elf-size
xtensa-lx106-elf-gcc-4.8.5     xtensa-lx106-elf-strings
xtensa-lx106-elf-gcc-ar        xtensa-lx106-elf-strip
xtensa-lx106-elf-gcc-nm
```

cefn, 2018-02-17:
Have you tried just running 

`PATH=/home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin/:$PATH`

...before the make commands, since I don't know if you have reloaded your .profile

The only reason for putting it in .profile is to make the change permanent, but it is only run on login.

lucianreynolds, 2018-02-17:
Yeah I exited out of vagrant and restarted it. I just ran the command and tried to make the axtls again. I just dont know what is going wrong here. Thank you so much for giving it some thought though.

cefn, 2018-02-17:
And if you run `echo $PATH` do you see the change is there?

There's no way a previous attempt to fix this has broken it is there? 

Does running `git status` from within the micropython-  repo and investigating a given file with `git diff filename` reveal any changes have been made relative to the original?

dpgeorge, 2018-03-07:
I assume this problem was solved.  Feel free to reopen the issue if not.

closed 2018-03-07 by dpgeorge