micropython/micropython #3610: round doesn't seem to be a special method

iyassou, 2018-02-13:
I'm writing a matrix module for Micropython and can't override the round special method and there are no references to it in tests/basics/special_methods.py and tests/basics/special_methods2.py.

Here's my round special method from my matrix class:
```
def __round__(self, _10_exponent=0):
    result = [[0 for i in range(self.size[0])] for j in range(self.size[1])]
    for i in range(self.size[0]):
        for j in range(self.size[1]):
            result[i][j] = round(self[i][j], _10_exponent)
    return matrix(*result)
```

Python 3.6.4 :
```
>>> a = matrix([1.1, 2.1, 3.1], [4.1, 5.1, 6.1], [7.1, 8.1, 9.1])
>>> a
matrix( [1.1, 2.1, 3.1],
        [4.1, 5.1, 6.1],
        [7.1, 8.1, 9.1] )
>>> round(a)
matrix( [1.0, 2.0, 3.0],
        [4.0, 5.0, 6.0],
        [7.0, 8.0, 9.0] )
```

Micropython 1.9.3 :
```
>>> a = matrix([1.1,2.1,3.1], [4.1,5.1,6.1], [7.1,8.1,9.1])
>>> a
matrix( [1.1, 2.1, 3.1],
        [4.1, 5.1, 6.1],
        [7.1, 8.1, 9.1] )
>>> round(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert matrix to float
```

dpgeorge, 2018-02-13:
Yes that's right, built-in `round()` does not defer to the `__round__` method, it just supports int and float at the moment.

To work around this, either use a method on the class (eg `matrix.round()`) or implement a custom `round()` method in Python that does defer to `__round__`.

iyassou, 2018-02-13:
Right then. I'll use a class method for this. Thanks.

closed 2018-02-13 by iyassou