micropython/micropython #3613: Trouble compiling for esp8266 port (make step); bunch of 'xyz.h' file(s) not found (#include xyz.h not found, starting with 'osapi.h')

soocki, 2018-02-14:
Compiler throws out errors, it cannot locate the '.h' files, the first one it complains about is 'osapi.h'.
FYI: I followed the instructions in ports/esp8266/README.md, the only doubt I have is that it mentions 'cd esp8266' but instead I did 'cd ports/esp8266' instead. The first two 'make' commands ('make -C mpy-cross' & 'make axtls') executed fine, the final 'make' doesn't seem to like me.
Help is very much appreciated.

pmp-p, 2018-02-14:
hi, osapi.h is in the sysroot of esp sdk 
cf: https://github.com/micropython/micropython/issues/3605

Maybe some one should make a front page big red warning about the fact that esp build uses cross-compilation.

soocki, 2018-02-14:
I have it in my path if that's what you are implying. At least I think that is what you are saying since you refernced that issue. Other than that I do not understand how is that supposed to help me. If I don't understand you correctly: my apologies since you are my only hope so far. I am not being ignorant, perhapsAnyway about the sdk-toolchain: I have compiled it in

pmp-p, 2018-02-14:
i did not test non-standalone mode and it is not the default so i guess your sysroot is not set.
you should retry with defautl standalone mode to see if include files are automatically found.

soocki, 2018-02-15:
Rebuilded the esp-open-sdk in standalone mode, it works. Appreciate the help. Ps. Do you think it is worth taking a note in 'ports/esp8266/README.md' about *NOT* building it in non-standalone mode for other doofuses like me?

closed 2018-02-15 by soocki