micropython/micropython #3615: esp32 port depends on stm32 and window ports' files

mfp20, 2018-02-14:
In order to build esp32 there's the need of:

ports/stm32/font_petme128_8x8.h
ports/windows/fmode.h
ports/windows/fmode.c

Those three files should be moved somewhere else, or the dependencies removed, in order to make the single port directory coherent. My advice is to move them:

extmod/font_petme128_8x8.h
mpy-cross/fmode.h
mpy-cross/fmode.c

Regards

dpgeorge, 2018-02-15:
Thanks for the suggestion.  Long ago there were many more dependencies between ports and most of those have since been fixed.

The font_petme128_8x8.h file probably belongs in extmod since it's used by extmod/modframebuf.c.  Maybe it should be moved to extmod/framebuf_petme128_8x8.h.

The fmode files are currently only used by mpy-cross but in principle can/should also be used in the windows port to make sure that writing files works correctly (see #2209).  So if they are to be move somewhere they should perhaps go in lib/.

mfp20, 2018-02-15:
On my way to create the patches. How can I deliver those? Attach patch files here? I already have a micropython fork repo for my own purposes but I'm doing something else, can't use that repo for pull requests.

mfp20, 2018-02-15:
Here they are: 
- [0001-ports-stm32-font_petme128_8x8.h-moved-to-extmod-font.patch.txt](https://github.com/micropython/micropython/files/1728795/0001-ports-stm32-font_petme128_8x8.h-moved-to-extmod-font.patch.txt)
- [0002-ports-windows-fmode.h-moved-to-lib-utils-in-order-to.patch.txt](https://github.com/micropython/micropython/files/1728796/0002-ports-windows-fmode.h-moved-to-lib-utils-in-order-to.patch.txt)
(had to rename them to .txt for github to attach, but they are git-format-patches).

mfp20, 2018-02-15:
Forgot to add the three files I modified to fix the paths (for fmode.h and fmode.c). Here it is

- [0003-fix-paths-for-new-fmode.-location.patch.txt](https://github.com/micropython/micropython/files/1728838/0003-fix-paths-for-new-fmode.-location.patch.txt)

dhylands, 2018-02-15:
The usual way is to follow the "fork and pull request" workflow documented here:
https://gist.github.com/Chaser324/ce0505fbed06b947d962

mfp20, 2018-02-15:
Thanks, but I really can't now. It takes a while for me to delete the one I have, fork again and make two branches: one for my project, one for upstream pull requests. I might be doing as soon as I start using micropython! But now I'm just packaging my framework (ie: https://github.com/mfp20/esp32-template-plus ). Could you be so kind to merge those patches ([and the ones I posted 4 minutes ago](https://github.com/micropython/micropython/issues/3618)) this time?

dhalbert, 2018-02-15:
You don't need to re-fork. Just make another clone of your own repo and create a new branch in that clone, starting from an earlier commit if necessary. Keep your threads of development separate in separate clones on separate branches.

dhylands, 2018-02-15:
You don't really need to delete anything. You create the fork on github, and you can then add a remote for the forked repository.

I normally clone the original repository, which winds up giving it a name of `origin`. I then point to my fork by doing:
```
git remote add dhylands git@github.com:dhylands/micropython.git
```
You can push a branch to your repository by doing:
```
git push dhylands branchname
```
I have remotes for several different people all in the same local repository. That way I can switch between different branches with just a single local repository.

dhylands, 2018-02-15:
For example, I currently have 4 different remotes:
```
895 >git remote -v
dhylands	git@github.com:dhylands/micropython.git (fetch)
dhylands	git@github.com:dhylands/micropython.git (push)
mcauser	https://github.com/mcauser/micropython.git (fetch)
mcauser	https://github.com/mcauser/micropython.git (push)
origin	https://github.com/micropython/micropython.git (fetch)
origin	https://github.com/micropython/micropython.git (push)
tralamazza	https://github.com/tralamazza/micropython.git (fetch)
tralamazza	https://github.com/tralamazza/micropython.git (push)
```

mfp20, 2018-02-15:
Ok, now I have 2 branches
```
$ git branch
* master
  mfp20
```
 and 2 remotes
```
$ git remote -v
mfp20	https://github.com/mfp20/micropython.git (fetch)
mfp20	https://github.com/mfp20/micropython.git (push)
origin	https://github.com/micropython/micropython.git (fetch)
origin	https://github.com/micropython/micropython.git (push)
```

How do I bind each of the branches with the right remote?

mfp20, 2018-02-15:
Ooookay... bloody you... I've been here banging my head on this monitor up to now. But seems to have been worth.

(just kidding, thanks for helping; I've been avoiding this lesson for about two years because every time I've thought  'ah-ah... I GOT YOU, GIT' ... I had to facepalm myself very soon, instead).

Please, have a look at https://github.com/mfp20/micropython . I have 3 branches now:
- mfp20-portsesp32removedeps (this issue and patches)
- mfp20-libsmod (issue #3618 )
- the first two merged into my own working branch esp32-template-plus (where I'll delete all the stuff unneeded to my project esp32-template-plus, like all the ports folders I don't need for esp32. Note: same name 'esp32-template-plus', two different entities: one is my project, one is the branch of my micropython fork that I want to pull stuff from, for my esp32-template-plus project)

And the web github interface advices to compare and create PRs for all of the three. I think I'm overshooting here... made the things more complicate than needed.

**Question1**: which one do I have to create a PR with? I'd go for both the first and second, so the micropython maintainer can choose whether to merge or not, any of the two, independently. But you said that I can push more commit to the same PR, no need to create one PR each commit. What do you mean? Which branch should I request for pushing upstream? In other words: which one should I rename to 'upstream' and use to send code to the main micropython repo?

dhylands, 2018-02-15:
What I do is create the feature branches off of master. I only create PRs for branches that I want the MicroPython maintainer to look at. For my own private stuff, I just ignore the UI.

If its been a while since I created the branch, I normally pull the latest master and rebase my branch. I make sure that I never make any changes on master since this always causes grief down the road.
```
git checkout master
git pull
git checkout my-feature-branch
git rebase master
git push -f remotename my-feature-branch
```

For me 'upstream' is 'origin'. What I meant about the commits is suppose I create the PR. And then Damien asks for changes. I just checkout the branch associated with the PR and make the changes, and then repush the branch. You'll then have a PR with 2 commits on it.

If you want you could also squash the commits into 1 or other things. If you rebase, or use commit --amend, then you'll need to use a forced push `git push -f remotename branchname`

mfp20, 2018-02-16:
> I make sure that I never make any changes on master since this always causes grief down the road.

Yep, don't worry, I'm used to work on mission critical systems holding sensitive data. That's why it took me so long to get to this point. I was trying to follow the two advices in a row I got 'to not delete and re-fork' but whatever I was doing I couldn't get a clean master, or couldn't delete a wrong branch pushed online, and so on ... so finally I deleted and re-forked, then pushed the changes in again, but not using master this time. I use master as downstream from this official repo, only.
Not that my stuff is sensitive and mission critical in this case but ... mine is a professional disease ... I even double check every piece of toilette paper before AND AFTER using it :)

Ok. Laugh done. Back to work.

I'm thinking ... "I gotcha GIT!!!" ... so I will merge the two issues into one single 'mfp20-upstream' branch, having 2 commits ahead of current master. Then PR on this branch, and leave it there for future PR to you micropython guys. You confirm to me that Damien will be able to (comfortably) select one, or the other one, or both the commits?

What about those? It's the log of esp32-template-plus branch after merging the two issues. Why the commits are three? I made two only, I swear!

```
commit 7997d63d13ada6ba8438b0ce5f77fd7ab6b674f7 (HEAD -> esp32-template-plus, mfp20/esp32-template-plus)
Merge: 2cbcac5c df79fefe
Author: mfp20 <20516565+mfp20@users.noreply.github.com>
Date:   Thu Feb 15 22:37:12 2018 +0100

    Merge branch 'mfp20-portsesp32removedeps' of https://github.com/mfp20/micropython into mfp20-esp32templateplus

commit df79fefef13a84db3a47415f001303dceebe4253 (mfp20/mfp20-portsesp32removedeps, mfp20-portsesp32removedeps)
Author: mfp20 <20516565+mfp20@users.noreply.github.com>
Date:   Thu Feb 15 22:19:12 2018 +0100

    ports/windows: moved fmode.h and fmode.c to lib/utils/ in order to remove ports/esp32 dependancy on ports/windows.
    
    Signed-off-by: mfp20 <20516565+mfp20@users.noreply.github.com>

commit 2cbcac5c738490a912175ee24678df580b018354 (mfp20/mfp20-libsmod, mfp20-libsmod)
Author: mfp20 <20516565+mfp20@users.noreply.github.com>
Date:   Thu Feb 15 19:59:35 2018 +0100

    ports/esp32: Makefile produces static libs. main.c is modified in order to not include app_main() in the static libs. The same old behaviour is kept by moving the old main.c into new files named esp32task.h and esp32task.c
    
    Signed-off-by: mfp20 <20516565+mfp20@users.noreply.github.com>

commit 73d1d20b46dda54340ad2819b865f47ca25f4850 (origin/master, origin/HEAD, master)
Author: Damien George <damien.p.george@gmail.com>
Date:   Thu Feb 15 16:50:02 2018 +1100
```

Thanks again for helping with git.

dhylands, 2018-02-16:
One of the commits is a merge commit (presumably you or the UI did a git merge?). Whenever I see a merge commit, it usually means I screwed something up

If you're proposing 2 different solutions for the same problem, then you should create 2 different PRs. Damien would then merge whichever one he likes and ignore the other.

While it is possible to select just 1 of 2 commits on the same branch, this would be highly unusual. You normally take all or none.

If I wanted to combine commits from 2 different branches, I normally create a new branch and then use git cherry-pick to grab the commits I want to combine.

mfp20, 2018-02-16:
> One of the commits is a merge commit (presumably you or the UI did a git merge?). Whenever I see a merge commit, it usually means I screwed something up

Yep, that's why I asked you to check. I've seen that kind of crap when I was rebuilding linux kernel's history back to 1996 using different repos. Can't remember the name of that feature but it's like clipping two single paper pages instead of having those connected to the same block note. And long term side effects scare me.

Anyway, my working branch can stay. But for the upstream branch I re-start from the two patch files, to avoid that crap. Then I PR that branch and cross my fingers.

How long does it take usually to see one little feature like that to appear in trunk? I don't mean to hurry anybody, It's just to know if it is worth to wait or not. Because I don't really need a branch for my app; when those two patches are mainlined, I'm set.

dhylands, 2018-02-16:
Whether it appears in trunk or not (and how long it takes) is entirely up to Damien.

mfp20, 2018-02-16:
Ahahahah ... 'It's ready when it's ready', the benevolent dictatorial developer said. Don't worry, that's my standard answer as well since 20 years. I asked for an Average Mean Time, or an estimate, or both. Whatever helps me to take my decisions :)

jonnor, 2023-07-30:
The merge request above seems to address the concerns raised in the initial report. So this issue can probably be closed?

dpgeorge, 2023-07-31:
`font_petme128_8x8.h` has been moved to `extmod/`.

The `fmode` files are still in `ports/windows/` but I think they are OK to stay there.  So, yes, let's close this issue.

closed 2023-07-31 by dpgeorge