micropython/micropython #3616: No dhcp_hostname parameter in ESP32 STA_config

Sunrise17, 2018-02-14:
In `modnetwork.c` of ESP32, there is no dhcp_hostname parameter described, but it consists in ESP8266. In due to this reason, i have received `ValueError: unknown config param` while using sta_if.config(dhcp_hostname=DEVICE_NAME) method.

ESP32 Version;
[image]

ESP8266 Version;
[image]

dpgeorge, 2018-02-15:
Thanks for the report, the `dhcp_hostname` option is indeed missing in the esp32 port.  I think implementing it would be as simple as calling `tcpip_adapter_set_hostname(self->if_id, hostname)`.

dpgeorge, 2018-02-19:
The 'dhcp_hostname' config option is now implemented by 27fa9881a9294c6a6875856c44101e5b33d27a3b

closed 2018-02-19 by dpgeorge