micropython/micropython #3618: esp32: static libs creation, patches attached

mfp20, 2018-02-15:
Currently the ports/esp32 Makefile isn't very flexible: it only creates a firmware.img file to be flashed on the esp32.

The attached patch introduces one more step between the .o files and the application.elf file, to create **static libs**:

[0001-ports-esp32-Makefile-produces-static-libs.patch.txt](https://github.com/micropython/micropython/files/1729022/0001-ports-esp32-Makefile-produces-static-libs.patch.txt)

**This allows for inclusion of the micropython vm into other C apps** (example: https://github.com/mfp20/esp32-template-plus ). As an example it is possible to use the required old esp-idf to build micropython, then drop the resulting libapp.a as a component of your project, and build your project using the last version of esp-idf sdk.

The old main.c code is moved to esp32task.h/esp32task.c files, where a new micropython_task function wants the core ID and returns a task handle. The old behaviour is preserved by creating a new app_main() in a separate new main.c file.

The resulting libesp32.a, libmp.a, and libapp.a libs have no app_main() function to avoid namespace conflicts. The application.elf includes the app_main() instead, in order to automatically run on esp32 (ie: same behaviour).

Please merge those changes.

dpgeorge, 2018-02-16:
> This allows for inclusion of the micropython vm into other C apps

The original idea is that MicroPython is the main app and you create add-ons (in C, C++, etc), rather than the other way around.  But I can see how it would be useful to be able to easily integrate MicroPython into some existing system.

> As an example it is possible to use the required old esp-idf to build micropython, then drop the resulting libapp.a as a component of your project, and build your project using the last version of esp-idf sdk.

I'm not sure this would work: wouldn't the MicroPython library depend on an older API and hence strange things happen when it tries to call newer versions of the functions that are in the main system it is imported into?

mfp20, 2018-02-17:
> The original idea is that MicroPython is the main app and you create add-ons (in C, C++, etc), rather than the other way around. But I can see how it would be useful to be able to easily integrate MicroPython into some existing system.

Well, you are producing a tool to open cans. But if someone finds it useful to stir the soup it's a better tool :)

If the tool come with a contract that forbids to use the tool for stirring the soup ... I simply don't buy it. I can write Lua, Haxe, Ruby, PHP, Perl, or simply C/C++. No point to lock myself into another walled garden.

> I'm not sure this would work: wouldn't the MicroPython library depend on an older API and hence strange things happen when it tries to call newer versions of the functions that are in the main system it is imported into?

That's the reason why I made it produce 3 static libs: 

- libesp32.a: "the API" mp was built with
- libmp.a: mp itself
- libapp.a: both

Then the user choose what to do: open cans, stir soup, "do nothing".

Including libmp.a only, using the newer sdk to build the rest could work. Anyway, that's not really an issue, because the original behaviour of your product is preserved; and there are no drawbacks: same resources, more flexibility, very little change in the code base.

pmp-p, 2018-02-17:
Thank you, Python tastes even better when embedded.

stinos, 2018-02-18:
> The attached patch

Could you just create Pull Request please? It allows others to easily see changes without having to download a file first. Plus it's what the maintainers prefer (https://github.com/micropython/micropython/wiki/ContributorGuidelines#how-to-contribute)

mfp20, 2018-02-18:
> Could you just create a Pull Request please?

The PR #3620 is there since two days. Just click on the link above your last message. Two commits for another issue, one for this one.

jonnor, 2024-09-29:
Embedding MicroPython is a valid usecase. There is now a hardware-independent way of embedding MicroPython - via the "embed" port. This is the recommended approach for embedding. See example at https://github.com/micropython/micropython/blob/master/examples/embedding/README.md

Note that while this functionality is minimally useful, is not in any way perfect - there is still work to be done. See the various open issues by searching for "embed" / "embedding".
But improving that port is generally the preferred way forward, over an ESP32 specific approach.

closed 2024-09-29 by jonnor