micropython/micropython #3619: Allow pausing and resuming MicroPython execution

Ecco, 2018-02-15:
Currently running `mp_call_function_0` will never return until execution is finished. This usually isn't a big deal, except in two cases:

* First, when running in a callback-based environment such as Emscripten that *needs* to return every now and then, otherwise blocking the whole browser
* Second, if you need to implement "input()" on top of a callback-based API

It would be very nice to allow "mp_call_function_0" to actually return and then resume execution later on. If I understand things correctly this should be made possible/easier thanks to `MICROPY_STACKLESS_STRICT`.

dpgeorge, 2018-02-16:
> First, when running in a callback-based environment such as Emscripten that needs to return every now and then, otherwise blocking the whole browser

Yes, that does present a real problem (see #3575 which adds Emscripten/JavaScript support and doesn't really get around this issue).  But it's really quite a deep issue because it's a difference in the execution model.  In JavaScript your functions can't run for very long either, and so you should only write short functions that don't block.  So MicroPython running in JavaScript via Emscripten would also need to have short-running functions, so that a call to, eg, `mp_call_function_0` would return quickly.

> It would be very nice to allow "mp_call_function_0" to actually return and then resume execution later on. If I understand things correctly this should be made possible/easier thanks to MICROPY_STACKLESS_STRICT.

Unfortunately it's really very difficult to make it resumable.  Even in stackless mode there is still a lot of C stack used by the C code itself.  One would need to implement "generators" at the C level to save the entire state of the call tree, return, and then resume.

dpgeorge, 2018-02-18:
Thinking about this more, it *may* be possible in *some* cases to pause/resume execution.  If the outer function is bytecode, and bytecode is only calling bytecode (ie not calling built-in C functions which then may call bytecode again), and STACKLESS is enabled, and the VM has a way of saving and restoring its state in a structure, and there are special checks and cases for all these conditions, then it could be that execution can be paused between VM opcodes.

Ecco, 2018-02-18:
Thank you @dpgeorge for your feedback on this issue!

> So MicroPython running in JavaScript via Emscripten would also need to have short-running functions, so that a call to, eg, mp_call_function_0 would return quickly.

Well, yes and no. My understanding is that if a MicroPython "execution context" can be saved somewhere not on the stack and paused/resume, then it could be possible to have the interpreter premptively "pause" after a given number of processed instructions (and return). And then a JS callback (e.g. timer) could then resume uPy execution later on. This could fit the cooperative multitasking model JS requires.

> If the outer function is bytecode, and bytecode is only calling bytecode (ie not calling built-in C functions which then may call bytecode again),

I don't realize if this is a "big" if or not.

Anyway, all that being said, I understand this issue might not be relevant for the "core" uPy functionality (definitely this has very little to do with microcontrollers uPy was intended for), but I think we both agree that demoing uPy on the web is really useful :) Sure Emscripten's not the only way, but it's a great way.

By the way, speaking of Emscripten, using Emterpreter allows to bypass this issue altogether, but it's a **lot** slower than regular Emscripten.

dpgeorge, 2018-02-18:
> If the outer function is bytecode, and bytecode is only calling bytecode (ie not calling built-in C functions which then may call bytecode again),
> 
> I don't realize if this is a "big" if or not.

Actually, it's not that big a set of "if"s.  Quite a lot of cases will be bytecode as the main outer function, and then calling bytecode.  So there may be a case to try and implement this functionality to see how it would work in practice.

> I think we both agree that demoing uPy on the web is really useful

Yes!  And I think that #3575 already goes a low way to demo it, even with all the restrictions on the execution model.

> using Emterpreter allows to bypass this issue altogether,

I didn't know about Emterpreter... might need to take a look, maybe it can help solve some issues with garbage collection.

jonnor, 2024-09-19:
Is this still relevant in recent MicroPython? There is now a Webassembly port which seems to have gotten quite far, seemingly without this?

jonnor, 2025-03-03:
Proposing to close as wont-fix

rversteegen, 2025-03-04:
This missing feature is the one and only reason that I'm not using MicroPython for a script interpreter in a game engine. It makes it impossible to support waits and real coroutines (i.e. without converting every single user and engine function into async), unlike language implementations like PUC Lua, LuaJIT, Squirrel, Stackless CPython, or CPython+greenlet. I've been watching this issue for years and considered implementing it myself.

I know MicroPython is not designed to be embedded as a script interpreter (neither was CPython and boy does it show). But it's such a fantastically complete and tiny Python implementation that it's a pity that it seems to have no ambition to fill a (very, very large) niche for which it could be a top choice (with further work to improve a nice API).