micropython/micropython #3621: webrepl crash

ds2k5, 2018-02-16:
Hello,
wann run the webrepl, but it crash if I open the URL with Bowser.
```

WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
connecting to network...
Network configuration: ('192.168.1.246', '255.255.0.0', '10.180.1.1', '212.87.130.92')
WebREPL daemon started on ws://192.168.1.246:8266
Started webrepl in normal mode

MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>> 
WebREPL connection from: ('192.168.1.114', 58848)
Traceback (most recent call last):
  File "webrepl.py", line 42, in accept_conn
  File "websocket_helper.py", line 34, in server_handshake
OSError: Not a websocket request

```

Here is the code:  **
```

import machine
import socket
import webrepl

def do_connect():
    import network

    SSID = 'MyWIFI'
    PASSWORD = 'MyWiFIPassword'

    sta_if = network.WLAN(network.STA_IF)
    ap_if = network.WLAN(network.AP_IF)
    if ap_if.active():
        ap_if.active(False)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect(SSID, PASSWORD)
        while not sta_if.isconnected():
            pass
    print('Network configuration:', sta_if.ifconfig())

do_connect()

webrepl.start()
```

```
**sudo ampy --port /dev/ttyUSB0 ls**
boot.py
webrepl_cfg.py
main.py  ** there is the CODE

```

sudo ampy --port /dev/ttyUSB0 get webrepl_cfg.py

show the password I set

What did I wrong ?

Thanks

dpgeorge, 2018-02-16:
Which browser/OS are you using?  You can try a different one to see if that is the issue.

ds2k5, 2018-02-16:
Manjaro Linux, Firefox 58.0.2

cefn, 2018-02-16:
Strikes me that you might be typing...

http://192.168.1.246:8266

...into your browser, rather than loading the WebRepl webpage from http://micropython.org/webrepl/ then switching your desktop's wifi to the ESP8266 Access point (password is micropythoN) and clicking on the Connect button to trigger the connection from within the page.

The WebREPL page is typically served from a different machine altogether, not on port 8266 of the device. Port 8266 is reserved for a websocket connection triggered by Javascript from within the WebREPL page.

If it irritates you to switch between AP and internet to be able to load/reload the WebREPL page then you can run your own copy of the WebREPL server locally on your desktop ( see https://github.com/micropython/webrepl ) . 

If for some reason you need the WebREPL to be actually hosted by the ESP8266 this is feasible, but non-standard ( see https://github.com/micropython/micropython/issues/2299#issuecomment-264750573 ).

Out of interest, does the fact there are two reports of WebREPL starting mean that it is actually started twice (perhaps in boot.py as well as later in your main.py script). Not sure what this would mean for stability.