micropython/micropython #3622: SD card instructions do not work with latest os module

skorokithakis, 2018-02-16:
The SD card instructions in the [sdcard.py file](https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py) for the ESP8266 seem out of date. `os.umount` takes one argument and `os.VfsFat` takes one two, but I can't figure out which ones they are. There's nothing about either object in the docs, and I can't find the `VfsFat` in the source either.

dpgeorge, 2018-02-16:
Yes those instructions are quite dated.  It should be something like:
```python
sd = ...
os.umount('/')
os.mount(sd, '/')
os.listdir()
```
If that works for you then we can update the text.

skorokithakis, 2018-02-16:
Unfortunately it doesn't :/ I tried this:

```python
import machine, sdcard, os
spi = machine.SPI(1)
sd = sdcard.SDCard(spi, machine.Pin(15))
os.umount("/")
vfs = os.VfsFat(sd)
os.mount(vfs, "/")
os.listdir("/")
```

and this:

```python
import machine, sdcard, os
spi = machine.SPI(1)
sd = sdcard.SDCard(spi, machine.Pin(15))
os.umount("/")
os.mount(sd, "/")
os.listdir("/")
```

but I get an exception after each mount step:

```
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
OSError: [Errno 19] ENODEV
```

My SD card (and SD object) works fine, I can get the sector count, read data, etc.

skorokithakis, 2018-02-16:
@dpgeorge Do you know if it would be easier to get this working with an ESP32, perhaps? I don't know how the SD card support is there.

skorokithakis, 2018-02-16:
Actually, that code worked, it was my SD card that was the problem. Thank you!

dpgeorge, 2018-02-18:
Ok, good that it works.  I updated the driver comment in 3759aa2cc98c96b869f3db5c4ac51778d3726669 (it's a bit different to what I wrote above).

> Do you know if it would be easier to get this working with an ESP32, perhaps?

The same code should work on the ESP32.

closed 2018-02-18 by dpgeorge