micropython/micropython #3632: ESP8266 with SDCard - AttributeError: 'module' object has no attribute 'SD'

ds2k5, 2018-02-23:
Hello,
I use a ESP8266 ( NodeMCU V3 )
https://cdn.instructables.com/FCL/V0OG/IVO7W206/FCLV0OGIVO7W206.LARGE.jpg
and a SDCardReader from LCtech-inc.com.
https://cdn.instructables.com/F6V/F1OJ/H5ENQ08B/F6VF1OJH5ENQ08B.LARGE.jpg

wire connections:

**ESP8266             SDCardReader**

GND                       GND

3V                          +3.3 V

D5 (PIN14)             SCL (CLK)

D7 (PIN13)             MOSI (CMD)

D6 (PIN12)             MISO (DAT0) 

Code:

```
import machine
import os
import uos
# clk cmd and dat0 pins must be passed along with
# their respective alternate functions
sd = machine.SD(pins=('GP14', 'GP13', 'GP12'))
os.mount(sd, '/sd')
# do normal file operations

uos.chdir(/sd)
uos.getcwd()
uos.listdir
uos.stat(/sd)
```

What did I wrong ?

Thanks

dpgeorge, 2018-02-23:
There is no `machine.SD` class on the esp8266.  Instead, try using the driver provided here: https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py

ds2k5, 2018-02-23:
Thanks dpgeorge,

I copied the file to my ESP8266
But did not understand how to CONNECT the wire correct
that the EXAMPLE from the CODE works

"
    import machine, sdcard, os
    sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))
    os.mount(sd, '/sd')
    os.listdir('/')
"

https://dziadalnfpolx.cloudfront.net/blog/wp-content/uploads/2015/09/esp8266-nodemcu-dev-kit-v3-pins.jpg

```
ESP8266      SDCARD
D5 (PIN14)    CS
SC                 MOSI
S0                  MISO
SK                  SCK
V3                  V3.3
GND              GND

```

mcauser, 2018-02-24:
If you are calling it like this:
`sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))`
Your CS pin needs to connect to D8/GPIO15.

This works on my Wemos D1 Mini:
ESP8266         | SD card reader
--------------- | --------------
D5 GPIO14 CLK   | CLK
D6 GPIO12 MISO  | MISO
D7 GPIO13 MOSI  | MOSI
D8 GPIO15 SS    | CS
3V3             | 3V3
GND             | GND

The nodemcu dev screenshot shows two SPI interfaces. The first one is for the ESP8266 onboard flash and the second is commonly referred to as HSPI or Hardware SPI and it's used for external peripherals. Connect to the HSPI pins and it should work on your board.

ds2k5, 2018-02-24:
Thanks mcauser, I wil try middle next week when I have access to the Board an the SDCard Reader

ds2k5, 2018-03-07:
Hello  mcauser,

Code:```
import machine
import os
import uos
# clk cmd and dat0 pins must be passed along with
# their respective alternate functions
sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))
os.mount(sd, '/sd')
# do normal file operations

uos.chdir(/sd)
uos.getcwd()
uos.listdir
uos.stat(/sd)
```

paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import os
=== os.listdir()
=== 
['boot.py', 'sdcard.py']

Import (STRG+E) the CODE (above)

Traceback (most recent call last):
  File "<stdin>", line 11
SyntaxError: invalid syntax

Line: 11 is:  uos.getcwd()

dpgeorge, 2018-03-07:
@ds2k5 on line 12 you'll need to put parenthesis after uos.listdir.  Also, the /sd needs to be in quotes for the chdir and stat.

ds2k5, 2018-03-08:
Hello dpgeorge,
thanks. But I did somethin wrong

paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import machine
=== import os
=== import uos
=== # clk cmd and dat0 pins must be passed along with
=== # their respective alternate functions
=== sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))
=== os.mount(sd, '/sd')
=== # do normal file operations
=== 
=== uos.chdir("/sd")
=== uos.getcwd()
=== uos.listdir()
=== uos.stat("/sd")
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
NameError: name 'sdcard' is not defined
>>>

mcauser, 2018-03-08:
You need a `import sdcard` after `import uos`

ds2k5, 2018-03-08:
Thanks a Lot to all!

```

import machine
import os
import uos
import sdcard 

# clk cmd and dat0 pins must be passed along with
# their respective alternate functions
sd = sdcard.SDCard(machine.SPI(1), machine.Pin(15))
os.mount(sd, '/sd')
# do normal file operations

uos.chdir("/sd")
uos.getcwd()
uos.listdir()
uos.stat("/sd")
```

closed 2018-03-08 by ds2k5