micropython/micropython #3637: unix build is broken on ubuntu 16.04 with MPY modules/upip_utarfile.py Segmentation fault (core dumped)

zoobab, 2018-02-24:
Hi,

I am trying to compile micropython on ubuntu 16.04, but I end up with a segfault:

```
root@52751b90a888:/mnr/ports/unix# make    
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
CC ../../py/mpstate.c
CC ../../py/nlr.c
CC ../../py/nlrx86.c
CC ../../py/nlrx64.c
CC ../../py/nlrthumb.c
CC ../../py/nlrxtensa.c
CC ../../py/nlrsetjmp.c
[...]
CC ../../extmod/vfs_fat_diskio.c
CC ../../extmod/vfs_fat_file.c
CC ../../extmod/utime_mphal.c
CC ../../extmod/uos_dupterm.c
CC ../../lib/embed/abort_.c
CC ../../lib/utils/printf.c
GEN build/frozen.c
CC build/frozen.c
MPY modules/upip_utarfile.py
../../py/mkrules.mk:115: recipe for target 'build/frozen_mpy/upip_utarfile.mpy' failed
make: *** [build/frozen_mpy/upip_utarfile.mpy] Segmentation fault (core dumped)
```

Any idea how to fix it? Can you reproduce?

zoobab, 2018-02-24:
Same error under fedora27.

peterhinch, 2018-02-25:
Builds OK under Mint 17.1. Did you guys rebuild the cross-compiler?

dpgeorge, 2018-02-26:
Please try doing `make clean` in mpy-cross and ports/unix, then recompile both of those components.

dpgeorge, 2018-04-26:
I assume this was resolved.  If not feel free to reopen.

zoobab, 2018-04-26:
Still the same with the latest commit (c7818032b160961bbd88a57926a78fcab297e4f2), I will try to make a Dockerfile so that you have see it by yourself.

hroncok, 2019-01-09:
Hi. As a Fedora maintainer, I had to disable build on ix86 due to failures. The build builds fine on x86_64 or arm. There's a segmentation fault during build on ix86.

```
MPY modules/upip_utarfile.py
mkdir -p build/frozen_mpy/
../../mpy-cross/mpy-cross -o build/frozen_mpy/upip_utarfile.mpy -s upip_utarfile.py -mcache-lookup-bc modules/upip_utarfile.py
BUILDSTDERR: make: *** [../../py/mkrules.mk:114: build/frozen_mpy/upip_utarfile.mpy] Segmentation fault (core dumped)
```

[Full build.log](https://github.com/micropython/micropython/files/2740354/build.log.txt)

dpgeorge, 2019-01-11:
@hroncok from your build log it looks like you are building v1.9.4.  Can you please try building the very latest code from master branch of this repository?

dpgeorge, 2019-01-13:
> There's a segmentation fault during build on ix86.

I can confirm this segmentation fault on Fedora 29 using MicroPython v1.9.4.  The issue is in nlrx86.c:nlr_push().  It seems that the calling convention on this system does not have a function prelude.

The problem seems to be fixed by c97607db5ccc03afbccacf853f2cd06305c28251, at least for me, because the Fedora 29 installation I tried had gcc 8.2.1 installed, which supports naked functions on x86.  The bug is likely still there for installations with older gcc (less than version 8).

dpgeorge, 2019-01-13:
@zoobab to check if the issue you had is the same as @hroncok, can you please confirm you used an x86 based system (32-bit)?

hroncok, 2019-01-13:
Thanks. I'll try on master and/or on 1.9.4 with that commit applied.

hroncok, 2019-01-13:
c97607d fixes this for me.

closed 2021-07-22 by dpgeorge