micropython/micropython #3638: After a while, Uasyncio server stops showing Json Http Body in Request ( ESP32-micropython )

Sunrise17, 2018-02-25:
Dear @dpgeorge, I am posting Json data from Swift App to ESP32 (micropython installed with uasyncio). After a while, it stops showing Json Body in the received request as you can see in attached debug below. From my same mobile app, i have sent same data to Python Server on PC (BaseHTTPServer installed), it is receiving JSON data without any problem. I will be glad if you can let me know about your opinions.

**Code in ESP32;**
    import time
    import json
    import uasyncio as asyncio
    
    async def handle_client(reader, writer):
        req = await reader.read()
        print("req:: %s" % req)
        (headers, js) = req.decode().split("\r\n\r\n")

        await writer.awrite("HTTP/1.0 200 OK\r\n\r\n'Color: JSON'\r\n")
        await writer.aclose()
    
    loop = asyncio.get_event_loop()
    loop.create_task(asyncio.start_server(handle_client, "192.168.1.71", 5700))
    loop.run_forever()

**Code in Swift;**

    static func sendPostToServer(json: [String: Any], ipAddress: String) {
        let jsonData = try? JSONSerialization.data(withJSONObject: json)
        // create post request
        let url = URL(string: "http://\(ipAddress):5700")!

        var request = URLRequest(url: url)
        request.httpMethod = "POST"
        request.setValue("application/x-www-form-urlencoded", forHTTPHeaderField: "Content-Type")
        request.setValue("application/json", forHTTPHeaderField: "Accept")
        request.setValue("", forHTTPHeaderField: "Accept-Language")
        request.setValue("Chroma", forHTTPHeaderField: "User-Agent")

        // insert json data to the request
        request.httpBody = jsonData
        
        let task = URLSession.shared.dataTask(with: request) { data, response, error in
            guard let data = data, error == nil else {
                print(error?.localizedDescription ?? "No data")
                return
            }
            let dataString = String(data: data, encoding: .utf8) ?? ""
            print("Response: \(dataString)")
        }
        task.resume()
    }

DEBUG in ESP32;
req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[8,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[8,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[5,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[9,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[10,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[11,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n{"SLIDER":[12,[255,116,0]]}'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[13,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[14,[255,116,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n**{"SLIDER":[12,[239,255,0]]}'**

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 27\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

req:: b'POST / HTTP/1.1\r\nHost: 192.168.1.71: 5700\r\nContent-Type: application/x-www-form-urlencoded\r\nConnection: keep-alive\r\nAccept: application/json\r\nUser-Agent: Chroma\r\nAccept-Language: \r\nContent-Length: 26\r\nAccept-Encoding: gzip, deflate\r\n\r\n'

dpgeorge, 2018-05-02:
This might be fixed by 6681eb809a815a049ca069d522b9ad07548db535; please retest with the very latest esp32 code.

Sunrise17, 2018-07-01:
Unfortunately, after i have sent several request, server started not to receive json data. Especially when it started to do like that, i am resetting the board, then it becomes fine. But after it happens again. May be it can caused by the cache issue?

closed 2022-06-23 by dpgeorge