micropython/micropython #3642: NodeMCU ESP32 Failed: rst:0x10  RTCWDT_RTC_RESET

KISSMonX, 2018-02-26:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
````

pmp-p, 2018-02-26:
@KISSMonX, did you erased flash before loading firmware ?

dpgeorge, 2018-02-27:
How exactly did you download the firmware to the ESP32?  Note that you need to flash it to the location starting at 0x1000.

KISSMonX, 2018-02-28:
Thanks, @dpgeorge 
 using `esptool.py` download the esp32xxx.bin.
It works.  But, uPyCraft download not work. maybe not start 0x1000. i will post new issue to uPyCraft. 

```shell
$ esptool.py --chip esp32 --port COM3 write_flash -z 0x1000 esp32-20180226-v1.9.3-374-gc5fe610b.bin     esptool.py v2.2.1
Connecting........___
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 973392 bytes to 604013...
Wrote 973392 bytes (604013 compressed) at 0x00001000 in 53.7 seconds (effective 145.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

closed 2018-02-28 by KISSMonX