micropython/micropython #3644: Descriptor Support on Official builds?

cefn, 2018-02-27:
I note in https://github.com/micropython/micropython/wiki/Differences

> simplified support for descriptors is now implemented, as an opt-in feature

How can I opt-in to descriptors working? There are no obvious build flags called anything DESCRIPTOR from a github search of the repo.

I have been trying to port some CircuitPython libraries to run on Micropython, and for example [adafruit_register](https://github.com/adafruit/Adafruit_CircuitPython_Register) relies on descriptors to tersely define values dynamically populated from I2C register reads. 

My guess is that future Adafruit driver development is going down this track, so being able to run unmodified CircuitPython libraries might rely on descriptors or gnarly monkey-patching hacks to work around the absence of descriptors.

If it's a build flag, is it likely to be activated on e.g. Pyboard, ESP8266, ESP32 official builds soon?

Currently I have this on MicroPython v1.9.3 on 2017-11-01; PYBv1.1 with STM32F405RG following https://github.com/micropython/micropython/blob/da34b6ef452514170e8ce1d1819070a920c2568e/tests/basics/class_descriptor.py as expected if the support wasn't turned on...

```python
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== 
=== class Descriptor:
===     def __get__(self, obj, cls):
===         print('get')
===         print(type(obj) is Main)
===         print(cls is Main)
===         return 'result'
=== 
===     def __set__(self, obj, val):
===         print('set')
===         print(type(obj) is Main)
===         print(val)
=== 
===     def __delete__(self, obj):
===         print('delete')
===         print(type(obj) is Main)
=== 
=== class Main:
===     Forward = Descriptor()
>>> m = Main()
>>> m.Forward
<Descriptor object at 200032a0>

```

dpgeorge, 2018-02-28:
> Is it likely to be activated on e.g. Pyboard, ESP8266, ESP32 official builds soon?

The reason it's not enabled on most ports (it looks like only unix and windows have it enabled) is because it makes all attribute loads from a user instance very slow (due to an extra lookup needed per attribute).  See https://github.com/micropython/micropython/blob/master/py/objtype.c#L617-L618

So I'd rather not enable it....  If it can be shown that enabling it doesn't reduce performance by much, or if an optimisation can be found to improve its performance, then it would be interesting to enable.

cefn, 2018-03-02:
Would it be feasible to have a per-class and per-instance bitfield for objects marking the presence of attributes which might trigger special Python lookup or delegation (behaviours as per https://github.com/micropython/micropython/blob/master/py/objtype.c#L617-L618). This bitfield would be manipulated on class or instance having a named attribute set, rather than looking up all alternatives when retrieved. It would slow down all class or instance assignments by checking (at least) if the assigned name was 'dunder', or the value type was property() before proceeding with normal assignment logic, but future retrieval would be able to eliminate the possibility of having to consider any special behaviours on any class or instance attribute with a minimal bitmasking operation, and the bitmask might be able to steer which checks were actually needed (class or instance __getattr__, descriptor or property logic could each be indicated by a different set bit). The upshot would be that you would 'opt-in' to slowdowns on a per-instance and per-class basis, so if you chose to exploit property or descriptor logic, then the extra costs would apply to all your attribute lookups of that class or instance, but those who sought efficiency above all else would deliberately omit any use of the constructs and just see the cost of a bitmask test in attribute lookups. I speculate that in these cases, the bitmask-based logic would cause attribute lookup to be even faster than currently for all those classes and instances which deliberately omit special behaviours. Does this add up as an idea? Can the need for a lookup be eliminated by 'caching' this information in a bitfield on attribute assignment?

peterhinch, 2018-03-03:
A nice idea if it can be made to work.

It seems surprising that CicuitPython uses this performance-sapping abstraction for speed-critical operations like accessing I2C device registers. Descriptors are an obscure Python feature with weird constraints like having to be used at class rather than instance level. Novice Python coders are unlikely to be familiar with them. To me they increase the complexity, and reduce the readability, of device drivers.

tannewt, 2018-03-05:
@peterhinch Speed isn't the top priority for CircuitPython, instead we prioritize ease of use.

From the outside, data descriptors act like attributes. Attributes on objects are easy for beginners to use since they are similar to variables. Using attributes instead of functions also clarifies what is state versus behavior of a driver. The bulk of our users will only ever know them as attributes.

For the few people who do write drivers, we have [a register library](https://circuitpython.readthedocs.io/projects/register/en/latest/) that makes it easy to define attributes that mirror registers. [Here](https://github.com/adafruit/Adafruit_CircuitPython_DS3231/blob/master/adafruit_ds3231.py#L66) is a good example.

peterhinch, 2018-03-06:
Thanks for that interesting insight. BTW you guys are ace at documentation :)

Personally I'd be loath to see anything that saps general performance on the official Pyboard port.

tannewt, 2018-03-06:
No problem @peterhinch. Thanks!

Totally understand MicroPython performance on the pyboard is critical for you. MicroPython is great for that. :-) That's the beauty of having options.

nickovs, 2018-05-21:
I came across a need for this too today, for a similar reason to @cefn. Descriptors are certainly complex under the hood but they can make it much easier to present clear abstractions of configurable properties to the end user.

Has the performance difference of enabling descriptors been benchmarked on real-world code? A while back I was looking at the costs of various lookups. I didn't come to a clear conclusions about it but I did come up with an approach for a speedup. At the moment there are a dozen or so calls to `mp_load_method_maybe()` that use an `MP_QSTR__` constant as the key. Perhaps there could be a fast-fail version of `mp_load_method_maybe()` that takes a small `enum` value for the key and maintains a lazy-generated bit field indicating the absence of the enumerated keys. Changes to the `locals_dict` could clear the flags and failed attempts to look up the key would set the relevant bit, so that future calls would fail fast. This would reduce to cost of looking up `__get__` or `__next__` or `__init__` (to name a few that are used) to a check of a bit in a bitfield in the vast majority of cases. For the objects that do have a `__get__` you still need to find it, but if you're actually using descriptors then you probably don't mind.

nickovs, 2018-05-22:
To answer my own question, I fired up pystone on an ESP32 with and without descriptor support enabled. The performance impact as it stands is about 22%; on the tagged v1.9.4 release without descriptor support I get about 1052 pystones/second but enabling descriptor support this drops to 821 pystones/second. As a result, much as I would love to have descriptors enabled, I agree that there is probably too big a hit to do it by default.

dpgeorge, 2018-05-22:
> Has the performance difference of enabling descriptors been benchmarked on real-world code?

I just tested on a PYBv1.0 using v1.9.4-72 (20b4b85f7266ef2edbe492829585a14118a85d7a) and, surprisingly, it only gives a 0.3% performance reduction on pystone (10k passes), 1690.9 down to 1685.48.  If that is indeed true then I have no problem enabling it for stm32. 

But we do in general lack a good set of benchmarks and it would be nice to test descriptor support with a more varied set of benchmark tests.

> I fired up pystone on an ESP32 with and without descriptor support enabled.

Inspired, I also checked ESP32 (a Wrover-Kit) myself: using v1.9.4-72 I get a consistent 886.132 pystones/sec *without* descriptors, and a consistent 1297.52 pystones/sec *with* descriptors!  I think this shows that the ESP32 is very sensitive to caching effects, there is no other explanation.  In addition, that means that any given change to the ESP32 code (be it related to performance or not) can have a huge impact on actual performance of this device.

dpgeorge, 2018-05-22:
Running CPythons `bm_chaos.py` benchmark (with small mods to make it work with uPy on baremetal; see https://github.com/python/performance), I get a 1.25% performance reduction on PYBv1.0 with descriptors enabled, and a 15% performance improvement on ESP32 with descriptors enabled.

nickovs, 2018-05-22:
> I just tested on a PYBv1.0 using v1.9.4-72 (20b4b85) and, surprisingly, it only gives a 0.3% performance reduction on pystone (10k passes), 1690.9 down to 1685.48.
...
> Running CPythons bm_chaos.py benchmark [...] I get a 1.25% performance reduction on PYBv1.0 with descriptors enabled, and a 15% performance improvement on ESP32 with descriptors enabled.

Well, I'm glad I asked the question. If these results hold then it would be great to throw the switch on descriptors by default.

dpgeorge, 2018-05-22:
> If these results hold then it would be great to throw the switch on descriptors by default.

Ok, so the plan would be to get a decent subset of the CPython benchmarks running on uPy and use them to quantify how much the performance changes on the various bare-metal ports.

dpgeorge, 2018-05-30:
@nickovs I implemented an idea similar to your idea above of having a flag which is used to indicate whether properties/special accessors need to be looked up.  Please see #3826.

dpgeorge, 2018-06-08:
Now that it's efficient, descriptors were enabled on stm32, esp8266 and esp32 in 93150a0d40a6f29907d53bba8a4a44f1c5fb2449

closed 2018-06-08 by dpgeorge

mattytrentini, 2019-02-13:
Sorry to comment on a closed issue but does that mean we should remove (or strike out) point 4 in the [Differences By Design](https://github.com/micropython/micropython/wiki/Differences) with CPython?

dpgeorge, 2019-02-13:
> Sorry to comment on a closed issue but does that mean we should remove (or strike out) point 4

I updated it to reflect the current state.  It's still true that descriptors are not used fundamentally in uPy, but they are supported for user defined classes.