micropython/micropython #3645: no mp_const_empty_list_obj

adritium, 2018-02-27:
Is there supposed to be a lack of "empty list" object?

```
// Constant objects, globally accessible
// The macros are for convenience only
#define mp_const_none (MP_OBJ_FROM_PTR(&mp_const_none_obj))
#define mp_const_false (MP_OBJ_FROM_PTR(&mp_const_false_obj))
#define mp_const_true (MP_OBJ_FROM_PTR(&mp_const_true_obj))
#define mp_const_empty_bytes (MP_OBJ_FROM_PTR(&mp_const_empty_bytes_obj))
#define mp_const_empty_tuple (MP_OBJ_FROM_PTR(&mp_const_empty_tuple_obj))
#define mp_const_notimplemented (MP_OBJ_FROM_PTR(&mp_const_notimplemented_obj))
```

dpgeorge, 2018-02-28:
> Is there supposed to be a lack of "empty list" object?

Yes, it's intentional: lists are *mutable* so it doesn't make sense to provide a *constant* global one.  Use `mp_obj_new_list(0, NULL)` instead.

adritium, 2018-02-28:
(I’m on mobile right now)

I’m unpacking kw arguments using the mp function that takes in C-array and allows you to provide a default (mp_arg_t I think). 

If I want to provide an empty list as the default, how do I do that? 

Can I set the entry .u_obj = mp_obj_new_list(0, NULL)

adritium, 2018-02-28:
I guess I could use mp_const_none.

closed 2018-02-28 by adritium

dhylands, 2018-03-01:
Passing a mutable default argument is almost always a mistake. 
Read: http://docs.python-guide.org/en/latest/writing/gotchas/