micropython/micropython #3646: ussl.wrapsocket and mbedtls

tsc87, 2018-02-28:
The function arguments for ussl.wrapsocket for ESP32 do not match with the documentation. ussl.wrap_socket(sock, server_side=False, keyfile=None, certfile=None, cert_reqs=CERT_NONE, ca_certs=None)
It seems that the modussl_mbedtls file has a modified function that has parameters listed as "key" and "cert" and missing argument for ca_cert.
Also, in socket_new, the call to mbedtls_ctr_drbg_seed takes null_entropy_func instead of mbedtls_entropy_func. 
How to make the ussl.wrapsocket working for ESP32 for use with AWS IoT?

dpgeorge, 2018-03-05:
> It seems that the modussl_mbedtls file has a modified function that has parameters listed as "key" and "cert" and missing argument for ca_cert.

Yes there is a discrepancy with the docs and the implementation.  Actually, both the axtls and mbedtls implementations of ussl accept the same set of keyword args: key, cert, server_side, server_hostname.

The key and cert args are equivalent to keyfile and certfile respectively, except the former take a bytes object with the data rather than a filename to read the data from.  So you can use it like this:
```python
with open('mykeyfile', 'rb') as f:
    keydata = f.read()
with open('mycertfile', 'rb') as f:
    certdata = f.read()
sock2 = ussl.wrap_socket(sock, key=keydata, cert=certdata)
```

> Also, in socket_new, the call to mbedtls_ctr_drbg_seed takes null_entropy_func instead of mbedtls_entropy_func.

Yes that needs to be fixed.  But it should still work as-is.

Sunrise17, 2018-04-11:
Dear @dpgeorge , yes this is the same issue which i have opened in [Link](https://github.com/micropython/micropython/issues/3687)
When approx. will it be available for the Esp-WRoom-32 Boards? As far as i see, @danicampora applied for cc3200, allows passing through ca_certs in ssl_params. [~/ports/cc3200/mods/modussl.c](https://github.com/micropython/micropython/blob/d19e4f0ba4df487b4ebd36b5fe6a16e68c0afe77/cc3200/mods/modussl.c).

dpgeorge, 2018-07-10:
> Also, in socket_new, the call to mbedtls_ctr_drbg_seed takes null_entropy_func instead of mbedtls_entropy_func.

This was fixed by ea22406f7661edcce88defb9d20517ec967a5a9f

fatihak, 2018-09-14:
Hey @dpgeorge , has the wrap_socket function on the ESP32 micro python been fixed? I am trying to connect to AWS iOT but it doesn't work without the CA_Cert parameter.

m0ot, 2019-04-02:
As I newly understood, esp32 port supports the open source mbedtls and not axtls. However, I implemented the latest esp32 bin file https://micropython.org/download#esp32 on the esp32 board. Then I tried to speak secured mqtt connection using the ussl.wrap_socket function https://docs.micropython.org/en/latest/library/ussl.html but it speaks only TLS 1.2
I'm a bit confused! If esp32 bin file supports the open source mbedtls then it should speak with the server TLS 1.3 because mbedtls already supports the latest version (TLS 1.3).
Any ideas?

willcharlton, 2019-05-14:
I am also not sure what to make of this. According to [docs](https://github.com/micropython/micropython/blob/master/docs/library/ussl.rst), the `ussl` module should contain some constants for cert requirements, but when I look on my ESP32:

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.10.0', version='v1.10-342-g7359a9e2f on 2019-05-13', machine='ESP32 module with ESP32')
>>> import ussl
>>> dir(ussl)
['__class__', '__name__', 'wrap_socket']
```

No such constants are available.

Additionally, when `umqtt.simple.MQTTClient.connect()` calls [`.wrap_socket()`](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L61) on a socket object, I get complaints about unsupported arguments. 

Is there a way I can test which tls library I'm actually using during these attempts?

Is this a platform specific issue? (i.e. is this only present on ESP32 but not on other platforms)

Let me know if/how I can help. Supporting mTLS on devices is paramount for me. Thanks!

Molaire, 2021-07-23:
> 
> 
> I am also not sure what to make of this. According to [docs](https://github.com/micropython/micropython/blob/master/docs/library/ussl.rst), the `ussl` module should contain some constants for cert requirements, but when I look on my ESP32:
> 
> ```
> >>> import os
> >>> os.uname()
> (sysname='esp32', nodename='esp32', release='1.10.0', version='v1.10-342-g7359a9e2f on 2019-05-13', machine='ESP32 module with ESP32')
> >>> import ussl
> >>> dir(ussl)
> ['__class__', '__name__', 'wrap_socket']
> ```
> 
> No such constants are available.
> 
> Additionally, when `umqtt.simple.MQTTClient.connect()` calls [`.wrap_socket()`](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L61) on a socket object, I get complaints about unsupported arguments.
> 
> ```
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "lib/umqtt/simple.py", line 61, in connect
> TypeError: extra keyword arguments given
> ```
> 
> Is there a way I can test which tls library I'm actually using during these attempts?
> 
> Is this a platform specific issue? (i.e. is this only present on ESP32 but not on other platforms)
> 
> Let me know if/how I can help. Supporting mTLS on devices is paramount for me. Thanks!

Did you find an answer? I am facing the same problem with 1.16

daniel-redder, 2022-12-06:
I'm also having this issue. Any solutions?

bneigher, 2023-03-28:
Is there anyone who can provide an ETA on when this is going to be supported?  

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.19.1', version='v1.19.1 on 2022-06-18', machine='ESP32C3 module with ESP32C3')
```

ca_certs seems to not be a supported parameter for ssl_params in from the [micropython-mqtt](https://github.com/peterhinch/micropython-mqtt), specifically boiling down to  `ussl.wrap_socket` not supporting it.

```python
from umqtt.simple import MQTTClient

# Set up the AWS IoT certificate and key files
cert_file = "/cert/device.cert.pem"
key_file = "/cert/device.private.key"
root_ca_file = "/cert/root-CA.crt"  # Path to the root CA certificate file

with open(key_file, 'r') as f:
  PVT_KEY = f.read()
  
with open(cert_file, 'r') as f:
  CERT_KEY = f.read()

client = MQTTClient(client_id, endpoint, ssl=True, ssl_params={
    "cert": CERT_KEY,
    "key": PVT_KEY,
    # "ca_certs": root_ca_file,  # Specify the path to the root CA certificate file <-- need this
    "server_side": False,  # set this to False for client-side SSL
})
```

Carglglz, 2023-03-28:
@bneigher use `cadata`  as keyword argument instead of `ca_certs` see https://github.com/peterhinch/micropython-mqtt/blob/master/mqtt_as/README.md#9-the-ssl_params-dictionary, and this thread https://github.com/orgs/micropython/discussions/10559

bneigher, 2023-03-28:
thanks that seems to have fixed it.  I was also using a different library.

```python
import time
import machine
from mqtt_as import MQTTClient, config
from ubinascii import hexlify

client_id = hexlify(machine.unique_id()).decode('utf-8')

# Set up the AWS IoT endpoint details
endpoint = "xxxxxxx.iot.yyyy-east-1.amazonaws.com"
topic = "sdk/test/python"

# Set up the AWS IoT certificate and key files
cert_file = "/cert/device.cert.pem"
key_file = "/cert/device.private.key"
root_ca_file = "/cert/root-CA.crt"  # Path to the root CA certificate file

with open(key_file, 'r') as f:
  PVT_KEY = f.read()
  
with open(cert_file, 'r') as f:
  CERT_KEY = f.read()
  
with open(root_ca_file, 'r') as f:
  CA_KEY = f.read()

ssl_params = {
    'key': PVT_KEY,
    'cert': CERT_KEY,
    'cadata': CA_KEY,
    'cert_reqs': ssl.CERT_REQUIRED,
    'do_handshake':True
}
                    
# Connect to AWS IoT MQTT
def connectMQTT():
    print("MQTT CONNECTING")
    config['client_id'] = client_id
    config['server'] = endpoint
    config['ssl_params'] = ssl_params
    client = MQTTClient(config)
    client.connect()
    print("MQTT CONNECTED")
    return client

# Publish a message to the topic
def sendMQTT(client, message):
    client.publish(topic, message)
```

I'm seeing successful connections in aws, but my published messages are not showing up in aws.. onto the next adventure.

Carglglz, 2023-03-28:
@bneigher see also #9259, I think you need to indicate `server_hostname` in `ssl_params`, which in your case is `endpoint`
e.g.
```
{...
'server_hostname': endpoint }
```

bneigher, 2023-03-28:
no dice.. I'm actually noticing that `client.isconnected()` is False.. so maybe false alarm about it being truly connected