micropython/micropython #3647: ssl_handshake_status: -256

3bb, 2018-02-28:
I try to make some request using urequest module. My website is running on HTTPS ( using letsencrypt certificate ).  After ~250 requests I receive ssl_handshake_status: -256. 

Mycropython version : esp8266-20171101-v1.9.3.bin

dpgeorge, 2018-03-05:
> ssl_handshake_status: -256.

This means that the underlying socket connection was lost during setup of the SSL connection.  Maybe the network went down.  Or is this reproducible always at ~250 requests?

3bb, 2018-03-07:
Hi george,
apparently was a problem with my network but the problem is not solved.  Sometimes I received this error after ~1 minute even if the network is up and runnning.

This is how I made requests, is something wrong with my function ?

> import gc
> gc.enable()
> gc.collect()
> import urequests as requests
> gc.collect()
> 
> headersText = {
>     'Content-Type': 'application/x-www-form-urlencoded',
>     'Host': 'www.mywebsite.com'
> }
> 
> url = 'https://www.mywebsite.com/api'
> 
> key = 'foo'
> def postData(key):
>     data = 'parameter={!s}'.format(key)
>     data = b"%s" % data
>     try:
>         r = requests.post(url, data=data, headers=headersText)
>         response = r.json()
>         r.close()
>         del data
>         del r
>         return response
>     except:
>         print('postData Error')

Thank you !

ketsa, 2018-10-18:
Greetings, 

Not sure if related or if I should open a new issue.
I'm geting ssl-handshake-status -257 : 

`MicroPython v1.9.4-8-ga9a3caad0 on 2018-05-11; ESP module with ESP8266

>>> r = urequests.get("https://min-api.cryptocompare.com/data/pricemulti?fsyms=BTC,ETH,ENG,XMR,DOGE,BNB&tsyms=USD")

TLS buffer overflow, record size: 5168 (+5)
ssl_handshake_status: -257
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 112, in get
  File "urequests.py", line 100, in request
  File "urequests.py", line 60, in request
OSError: [Errno 5] EIO
`

dpgeorge, 2018-10-19:
@ketsa your issue is different, it is due to TLS buffer overflow (as mentioned in the output) and is a limitation of the esp8266 because it doesn't have much RAM.  See #3279 for ways around this.

sumpfralle, 2019-06-07:
I also experience this issue:
```shell
# printf 'from urllib.urequest import urlopen; urlopen("https://api.on-i.de/")' | micropython 2>/dev/null
ssl_handshake_status: -256
```

In my specific situation it is problematic, that the error message `ssl_handshake_status: -256` is written to stdout instead of stderr. This breaks the output formatting of my python script.

nirtz89, 2021-05-29:
Any updates on this topic?