micropython/micropython #3649: Error while Building micropython

nikhil1983, 2018-03-01:
Hey, 

I am currently trying to build and flash micropython on STM32F4DISC board. After first few steps of building,
In **micropython/ports/stm32/** 
when i execute the make and make BOARD=STM32F4DISC i get the following error. 

> > In file included from usbdev/core/inc/usbd_core.h:34:0,
>                  from usbd_conf.c:33:
> usbdev/core/inc/usbd_def.h:147:5: error: unknown type name 'uint8_t'
>      uint8_t   bmRequest;                      
>      ^
> usbdev/core/inc/usbd_def.h:148:5: error: unknown type name 'uint8_t'
>      uint8_t   bRequest;                           
>      ^
> usbdev/core/inc/usbd_def.h:149:5: error: unknown type name 'uint16_t'
>      uint16_t  wValue;                             
>      ^
> usbdev/core/inc/usbd_def.h:150:5: error: unknown type name 'uint16_t'
>      uint16_t  wIndex;                             
>      ^
> usbdev/core/inc/usbd_def.h:151:5: error: unknown type name 'uint16_t'
>      uint16_t  wLength;                            
>      ^
> usbdev/core/inc/usbd_def.h:158:3: error: expected specifier-qualifier-list before 'uint8_t'
>    uint8_t  (*Init)             (struct _USBD_HandleTypeDef *pdev , uint8_t cfgidx);
>    ^
> usbdev/core/inc/usbd_def.h:198:3: error: unknown type name 'uint8_t'
>    uint8_t *(*GetDeviceDescriptor)(struct _USBD_HandleTypeDef *pdev, uint16_t *length);
>    ^
> usbdev/core/inc/usbd_def.h:198:69: error: unknown type name 'uint16_t'
>    uint8_t *(*GetDeviceDescriptor)(struct _USBD_HandleTypeDef *pdev, uint16_t *length);
>                                                                      ^
> usbdev/core/inc/usbd_def.h:199:3: error: expected ';' before 'uint8_t'
>    uint8_t *(*GetStrDescriptor)(struct _USBD_HandleTypeDef *pdev, uint8_t idx, uint16_t *length);
>    ^
> usbdev/core/inc/usbd_def.h:205:3: error: unknown type name 'uint32_t'
>    uint32_t                status;
>    ^
> usbdev/core/inc/usbd_def.h:206:3: error: unknown type name 'uint32_t'
>    uint32_t                total_length;    
>    ^
> usbdev/core/inc/usbd_def.h:207:3: error: unknown type name 'uint32_t'
>    uint32_t                rem_length; 
>    ^
> usbdev/core/inc/usbd_def.h:208:3: error: unknown type name 'uint32_t'
>    uint32_t                maxpacket;   
>    ^
> usbdev/core/inc/usbd_def.h:214:3: error: unknown type name 'uint8_t'
>    uint8_t                 id;
>    ^
> usbdev/core/inc/usbd_def.h:215:3: error: unknown type name 'uint32_t'
>    uint32_t                dev_config;
>    ^

the error goes on.   
So how to remove this error ?
This type of problem didn't happened when i build the micropython few months before.

peterhinch, 2018-03-01:
Have you tried
```
make BOARD=STM32F4DISC clean
make BOARD=STM32F4DISC
```

dpgeorge, 2018-03-01:
I did see this in the Travis build on occasion but it magically fixed itself.  What toolchain are you using (eg arm-none-eabi-gcc, and newlib?).

nikhil1983, 2018-03-01:
@dpgeorge Yes i am using the arm-none-eabi-gcc

nikhil1983, 2018-03-01:
@peterhinch yes i tried make clean, and still the same problem exist.

robert-hh, 2018-03-01:
Since a while I have these errors in a few files, that the standard int types are unknown. It's easy to be fixed, by adding
`#include <stdtint.h>`
to the file in question. But it's strange, and may depend on a compiler version or other setting in you system.

nikhil1983, 2018-03-01:
i have included now this two files ""#include "stm32f4xx.h"
#include "stm32f4xx_hal.h" "" in **usbd_conf.h** file.  Its building properly now. Sir @dpgeorge just update this usbd_conf.h file so that no one face same issue.

dpgeorge, 2018-03-02:
@nikhil1983: can you please confirm if adding `#include <stdint.h>` to usbd_conf.h fixes it for you?  Like this:
```diff
--- a/ports/stm32/usbd_conf.h
+++ b/ports/stm32/usbd_conf.h
@@ -34,6 +34,7 @@
 #define __USBD_CONF_H
 
 /* Includes ------------------------------------------------------------------*/
+#include <stdint.h>
 #include <stdio.h>
 #include <stdlib.h>
 #include <string.h>
```
If it does fix it then I'll make the change.

dpgeorge, 2018-03-09:
Fixed by 993f4345c0fd57668984dd1248e84017c2352238

closed 2018-03-09 by dpgeorge

linj9iang, 2020-11-13:
i appear this error as:
cores\maple\avr\dtostrf.c:34:3: error: unknown type name 'uint8_t'      
   uint8_t negative = 0;
   ^
exit status 1