micropython/micropython #3650: SSL - Guru Meditation Error

dreasGh, 2018-03-02:
Hi,
i have an esp32 with the following version:
(sysname='esp32', nodename='esp32', release='1.9.3', version='v1.9.3-386-g9884a2c-dirty on 2018-03-02', machine='ESP32 module with ESP32')

I have a custom httpserver and umqtt client running in parallel with uasyncio.
When i call an API (exposed by my httpserver) wich trigger uasyncio.open_connect with ssl=True, my esp come to reboot with the following errors:

```
Warning: uasyncio SSL support is alpha
Traceback (most recent call last):
  File "httpserver.py", line 87, in run
  File "httpserver.py", line 181, in handle_request
  File "command_controller.py", line 131, in tlcs_call
  File "uasyncio.py", line 226, in open_connection
OSError: [Errno 12] ENOMEM

Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400dd306  PS      : 0x00060d30  A0      : 0x800dbf45  A1      : 0x3ffb6010  
A2      : 0x00000000  A3      : 0x3ffebba0  A4      : 0x3ffb61b0  A5      : 0x3f40d26c  
A6      : 0x000000db  A7      : 0x00000080  A8      : 0x3f409f50  A9      : 0x3ffb6000  
A10     : 0x00000000  A11     : 0x3ffebba0  A12     : 0x00000006  A13     : 0x3ffb61a0  
A14     : 0x3ffb6158  A15     : 0x3ffb61a0  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x400dd306:0x3ffb6010 0x400dbf42:0x3ffb6030 0x400dbfb9:0x3ffb6050 0x400e5ca0:0x3ffb6070 0x400ed1e1:0x3ffb60a0 0x400df9d5:0x3ffb60c0 0x400dbe01:0x3ffb60e0 0x400dbe45:0x3ffb6100 0x400dccf0:0x3ffb6130 0x400d34f8:0x3ffb61b0 0x400ef3f4:0x3ffb61f0 0x400e719f:0x3ffb6210 0x400df99a:0x3ffb6230 0x400dbe01:0x3ffb6250 0x400dbe69:0x3ffb6270 0x400e8df1:0x3ffb6290 0x400dfd88:0x3ffb6330 0x400dfdd9:0x3ffb6350 0x400dfe8c:0x3ffb6380 0x400dc0ce:0x3ffb63a0 0x400e709d:0x3ffb63d0 0x400df9d5:0x3ffb63f0 0x400dbe01:0x3ffb6410 0x400e8d65:0x3ffb6430 0x400dfb34:0x3ffb64d0 0x400dbe01:0x3ffb6500 0x400dbe69:0x3ffb6520 0x400e8df1:0x3ffb6540 0x400dfb34:0x3ffb65e0 0x400dbe01:0x3ffb6660 0x400dbe2e:0x3ffb6680 0x400f7f22:0x3ffb66a0 0x400f8179:0x3ffb6740 0x400ef31f:0x3ffb6760

Rebooting...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
```
With plain socket and SSL i have the same result.
The problem is ussl.wrap_socket(s) in both of case.
If i try to stop umqtt client the httpserver works fine.

Sorry for my bad english.
Thanks in advance.

peterhinch, 2018-03-03:
The ENOMEM error from MicroPython indicates that a RAM allocation failed, probably indicative of heap fragmentation. You might want to read [the section on the heap](http://docs.micropython.org/en/latest/esp8266/reference/constrained.html) for ways of dealing with this.

dreasGh, 2018-03-04:
Hi @peterhinch,

thanks for the reply.
I have followed the guidelines as micropython doc described before posting my problem.

After my application has been executed, below the result of micropython.mem_info(1):

 `stack: 2048 out of 15360
GC: total: 111168, used: 26160, free: 85008
 No. of 1-blocks: 406, 2-blocks: 55, max blk sz: 264, max free sz: 5306
GC memory layout; from 3ffe4db0:
00000: MDhhhDMMDDhDhDBBBBhhh===h===Dhh==h==============================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: =========================================hBhTLDhShBhTSMDhhMDh=hM
01400: BSBDDhh==SDBBBBBBBSh=h========DBBBhBShDh=======h================
01800: ================================================================
01c00: ================================================================
02000: =====================================================h===h====B=
02400: h======h==MDh==Dhh=h========BhDhDhh======hDBDh========hMDShhMDMh
02800: h===DhTLB=Bh=h===hDhBTBBBhBBBBBB=h===hDBBh=Dh========hSDDh=hDShh
02c00: SSDhShShhShh=======ShDh===hThSThShDShSSThh=hDh==Dhh==h===Shh====
03000: h===LSh===h=hh===h===hh====h====hhSDT=ShShT=ShShT=Shh=ShhhSDh===
03400: =======================================h========SB=BhSDBBh===hSh
03800: ===BSBh=h==h==h===hhh=h==hBSDBBh=hTTh===SSDBBDSLh===============
03c00: =h===hB=SB=DBLB=BBBBSSh===DSh==================h================
04000: ==h===SB=BhB=hBhBSDBhh===h===SBBhB=hBhBhhSB=h===h===B=SB=hLMDMhD
04400: h=SMhhh===SDhhh===h=Dhhh===Mh=DDh=ABhBBBLBBh=BhBDhB=h==hBhDhh===
04800: =====hBhh=BLBhBhh====hBLhBDhBLhLLhBLhhh....BDh=h===h==BDBh=h=hBD
04c00: h=Lhh=ShSBDh=h===============hSBDh=hLhShSBDh=h==SSB=BhSSh=BDh=h=
05000: ==========hh=h===h===========SDh==LDDhhh===Bh===================
05400: ============h==Dh=B==BSBhB=B=BBLhSShShh=========================
05800: ====hB=SB=BBh===h=Dh========hBBBhB=hSBhhhBhh====h===hShh=h...h==
05c00: ==h=============================================================
06000: ==============================================h==============h==
06400: =============h===============LhShShShh====......................
        (82 lines all free)
1b000: ....................................`

dpgeorge, 2018-03-05:
Each SSL socket takes a lot of RAM (maybe up to 32k each) so you can only have a small number of them open at any one time.  On the esp32 the memory for the SSL state comes from the RTOS heap, not from the uPy object heap.

@dreasGh how many SSL sockets do you have open at one time?  Do you make sure to close them when they are finished being used?

Ways to work around the problem would be: 1) only use one SSL socket at a time; 2) edit the code to decrease the amount of RAM used by the uPy GC, so there is more room for the SSL sockets; 3) get a module with external SPIRAM and enable SPIRAM support.

dreasGh, 2018-03-05:
Hi @dpgeorge,

I have one ssl socket opened for mqtt client (umqtt.robust lib) that runs forever (to check messages into queue) and another for httpserver, closed after each request.
If i try to run mqtt client without ssl socket everything works fine.

In details:

1.
client -> API call -> httpserver -> https call to google.it (ssl socket test) -> ssl socket closed -> client socket closed
mqtt_client -> ssl socket -> rabbitmq
After this ENOMEM raised and esp32 restarts.

2.
client -> API call -> httpserver -> https call to google.it (ssl socket test) -> ssl socket closed -> client socket closed
mqtt_client -> plain socket -> rabbitmq
After this everything works fine.

So, i think that two ssl socket can't live together.
Then points one and two are not reachable. Other than third point is there any options you could suggest?

Thanks for the help!

closed 2018-05-18 by dreasGh

arslan437, 2019-10-15:
I'm trying to connect esp32 with micro to azure port using http but i'm recving **ENOMEM**. I'm new to micro python so any can help me here is my code 

`s = socket.socket()
    ai = socket.getaddrinfo(host, 443)
    addr = ai[0][-1]
    s.close()
    gc.collect()
    count = 0
    errorCount = 0
    while True:
        # if not oledDisplay:
        #     builtinLed.value(0)  # turn on led

        checkwifi()
        newSasToken()
        count = count + 1

        #temperature, pressure, humidity = mySensor.measure()
        temperature = 37
        pressure = 52
        humidity = 70
        light = 0
        freeMemory = gc.mem_free()

        print (gc.mem_free())

        data = b'{"Id":%u,"Mem":%u,"Celsius":%s,"hPa":%s,"Humidity":%s,  "Light":%d, "Errors":%d}' % (
            count, freeMemory, temperature, pressure, humidity,  light, errorCount)
        print (gc.mem_free())
        
        print ('trying')
        s = socket.socket()
        print(addr)
        s.connect(addr)
        
        s = ussl.wrap_socket(s)  # SSL wrap
        print ('sent')
        # Send POST request to Azure IoT Hub
        # s.write("POST /devices/" + cfg.deviceId +
        #         "/messages/events?api-version=2016-02-03 HTTP/1.0\r\n")
        
        s.write("POST " + hub.endpoint + " HTTP/1.1\r\n")
        # HTTP Headers
        s.write("Host: " + host + "\r\n")
        s.write("Authorization: " + SAS + "\r\n")
        s.write("Content-Type: application/json\r\n")
        s.write("Connection: close\r\n")
        s.write("Content-Length: " + str(len(data)) + "\r\n\r\n")
        # Data
        s.write(data)

        
        # Print 128 bytes of response
        print(s.read(128)[:12])

        s.close()
        time.sleep(sampleRate)`

MrSurly, 2019-10-15:
@arslan437 

Please:
* Create a new issue
* Post your full code so the problem can be reproduced (e.g. `checkWifi` and `newSasToken` are missing)

Thanks.

arslan437, 2019-10-16:
@MrSurly 
Ok I will create a new issue. 
I'm using this git project without any change (sensors and i2c code is commented)
[https://github.com/gloveboxes/ESP32-Micropython-Azure-IoT-and-Event-Hub-Client](url)

arslan437, 2019-10-16:
@MrSurly 
I have created the new issue and pasted full code on that issue. here is the link of that issue 
[https://github.com/micropython/micropython/issues/5219](url)

I will be very grateful for your consideration,  look at code if you can and suggest me is there any way that we can resolve this issue or I have to upgrade hardware.