micropython/micropython #3651: esp32: latest prebuilt firmware hangs

tuupola, 2018-03-02:
Just a quick note. Latest prebuilt firmware (esp32-20180302-v1.9.3-386-g9884a2c7.bin) does not boot up. Older prebuilt firmwares I have downloaded previously work ok which should rule out a faulty board.

```
$ esptool.py --port /dev/cu.SLAB_USBtoUART  --baud 115200 --after no_reset erase_flash
esptool.py v2.2.1
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.1s
Staying in bootloader.

$ esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 esp32-20180302-v1.9.3-386-g9884a2c7.bin

esptool.py v2.2.1
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x022f
Compressed 973424 bytes to 604061...
Wrote 973424 bytes (604061 compressed) at 0x00001000 in 53.3 seconds (effective 146.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

$ screen /dev/tty.SLAB_USBtoUART 115200

<pressed reset here>

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:4436
load:0x40078000,len:0
load:0x40078000,len:11816
entry 0x4007a9fc
```

peterhinch, 2018-03-03:
I have installed and run today's build (esp32-20180303-v1.9.3-394-gadda38cf.bin) with:
```
esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000
```

dpgeorge, 2018-03-05:
I just tried the latest build from the download site (esp32-20180305-v1.9.3-397-g0acf868b.bin) and put it on an M5 Stack and it worked.  I did:
```
$ esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash
$ esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 esp32-20180305-v1.9.3-397-g0acf868b.bin
```

mattytrentini, 2018-03-22:
I just tried with today's build (esp32-20180322-v1.9.3-477-g7b0a020a.bin) with the _same hardware_ - **M5Stack w 4MB PSRAM** - and reproduced your failure @tuupola. 

Note that I had the same experience as @dpgeorge and @peterhinch when trying today's build on _non-PSRAM hardware_. I tried with a 'regular' M5Stack and 2x TTGO Mini 32's. Worked fine!

However I _was_ able to get it working on the 4MB PSRAM unit; it seems that one of your parameters was incorrect. The Loboris port, [for 4MB PSRAM devices](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/firmware/esp32_psram/flash.sh), suggests using `--flash_freq=40m` instead of `80m`. I changed that and it booted straight into the REPL:

```
$ esptool.py --port /dev/ttyUSB1 --chip esp32 erase_flash
esptool.py v2.1
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Hard resetting...

$ esptool.py --port /dev/ttyUSB1 --chip esp32 --baud 460800 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 ~/Downloads/esp32-20180322-v1.9.3-477-g7b0a020a.bin 
esptool.py v2.1
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 973600 bytes to 604294...
Wrote 973600 bytes (604294 compressed) at 0x00001000 in 14.6 seconds (effective 532.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

$ screen /dev/ttyUSB1 115200
>>> print('40m FTW')
40m FTW
```

Can you please re-test with those options @tuupola and let us know if that resolves the problem for you too?

mattytrentini, 2018-03-22:
Oh, assuming it does sort out your issue, you might want to update your [excellent blog post](https://appelsiini.net/2018/m5stack-esp32-firmware-cli/) @tuupola!

dpgeorge, 2018-05-18:
I assume this was resolved by using 40m.

closed 2018-05-18 by dpgeorge

tuupola, 2018-05-18:
80m works ok too with the latest.

```
$ esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 esp32-20180518-v1.9.4-37-g58331e3c.bin
esptool.py v2.3.1
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x022f
Compressed 1013984 bytes to 630194...
Wrote 1013984 bytes (630194 compressed) at 0x00001000 in 55.5 seconds (effective 146.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

$ screen /dev/cu.SLAB_USBtoUART 115200
>>> print("FOO")
FOO
```

dpgeorge, 2018-05-18:
> 80m works ok too with the latest

Ok, that's good to know.  It's probably something to do with the update ESP IDF, which is now v3+.