micropython/micropython #3652: Add mp_obj_make_copy(mp_obj_t)?

adritium, 2018-03-04:
I'm implementing the `.attr` function for my new type.

For my "load attribute" functionality, I am returning by copy and I am finding that a function `mp_obj_make_copy(mp_obj_t)` would be useful.

When I get an item using `mp_obj_dict_get()`, the type is `mp_obj_t`. 
If I want to make a copy, I have to call the appropriate "make_new" function.

Why not have `mp_obj_make_copy(mp_obj_t obj_in)` that calls the underlying "copy" functionality and returns `mp_obj_t`?

dpgeorge, 2018-03-07:
> Why not have mp_obj_make_copy(mp_obj_t obj_in) that calls the underlying "copy" functionality and returns a mp_obj_t?

I guess the reason is that such a function was not needed so far.  Note that CPython does have a module for this, the `copy` module, which can do shallow and deep copies.  Implementing the functionality in C would require handling lots of different cases for the different types.

adritium, 2018-03-07:
I agree it would require handling but why would it be so difficult? 
You'd check for the "basic" objects "small int", "qstr" and then you'd call `.make_new` for the other object types.

Is there a complication?

adritium, 2018-03-12:
I see that to make it deep copy, you'd have to modify each `mp_type*` to do a deep-copy.

For example, this creates a shallow copy:

```
mp_obj_t mp_obj_new_list(size_t n, mp_obj_t *items) {
    mp_obj_list_t *o = list_new(n);
    if (items != NULL) {
        for (size_t i = 0; i < n; i++) {
            o->items[i] = items[i];
        }
    }
    return MP_OBJ_FROM_PTR(o);
}
```

adritium, 2018-05-13:
@dpgeorge 

> I guess the reason is that such a function was not needed so far.

Now we [have a need](https://github.com/micropython/micropython/issues/3774#issuecomment-388573461) in #3774

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge