micropython/micropython #3655: micropython/tests/pyb/adc.py appears to have bugs

peterhinch, 2018-03-05:
This [test](https://github.com/micropython/micropython/blob/a5fb699d87ffa178736021c4763d323be54e87d4/tests/pyb/adc.py#L19) will surely always pass as a bytearray element can never exceed 255.

The error is repeated further down the code (L27, L32) when the assertion tests re-examine the contents of **buf** rather than the arrays.

The consequence is that the tests always pass. Unfortunately if you examine the actual arrays the values read (on the working Pyboard I'm using) often substantially exceed 500: I'm seeing values of over 1000. It's evident that the pull-downs are insufficiently strong for the purpose. Using pull-ups gives similar errors. (A wire link produces correct results).

Finally initialising buffers and arrays to zero seems not to be ideal as a completely dead ADC driver might leave the contents unchanged with the test still passing.

chuckbook, 2018-03-05:
Just a note: According to **DocID018909 Rev 12 pp 271** no internal pull up/down shall be active while a
pin is in analog mode. It may be better to use the internal temp sensor or VREFINT as source for this test.

peterhinch, 2018-03-05:
Yes. I've just spotted [this line](https://github.com/micropython/micropython/blob/a5fb699d87ffa178736021c4763d323be54e87d4/ports/stm32/adc.c#L196) which seems to do just that. So, if I'm reading the code correctly, the pullup is disabled when the ADC is instantiated. That might explain the odd measurements.

dpgeorge, 2018-03-06:
> After looking at this some more I can't see a way of testing ADC.read_timed() using Vref or temperature.

Might it be possible to do this by creating an ADC object from an integer, which forces the use of a certain channel?  Eg `adc = ADC(16)` should give you access to the temperature sensor.

peterhinch, 2018-03-06:
Yes, that works. I'll push a revised (and fixed) test program with https://github.com/micropython/micropython/pull/3656 once we've settled the design, if that's OK.

dpgeorge, 2018-03-06:
> if that's OK.

OK!

dpgeorge, 2018-04-23:
Fixed by 0096a4bd005606489a5b7fdfda8e2a60a1709e13

closed 2018-04-23 by dpgeorge