micropython/micropython #3657: After SyntaxError on import, cannot re-attempt import

tomlogic, 2018-03-06:
I ran into this working on our platform support code, and was able to reproduce it in Windows.

```
$ ./micropython
MicroPython v1.9.3-722-gb8461320-dirty on 2018-03-05; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import mini
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mini.py", line 8
SyntaxError: invalid syntax
>>> dir(mini)
['__name__', '__file__']
# fix errors in mini.py and try again
>>> import mini
>>> dir(mini)
['__name__', '__file__']
>>> del mini
>>> import mini
>>> dir(mini)
['__name__', '__file__']
```

If I exit the REPL and try again, it works fine.

tomlogic, 2018-03-06:
Without knowing a lot about how MicroPython manages modules, I'm seeing the following problem:

* `mp_obj_new_module()` creates an empty module inside of `mp_loaded_modules_dict`.
* `py/buildinimport.c` tries to compile the module, but fails.
* Stub entry remains in `mp_loaded_modules_dict`.

Would it make more sense to compile the module, and only add it to `mp_loaded_modules_dict` if successful, maybe via `mp_module_register()`?

I'll create a branch off of HEAD tomorrow, but here's my proposed patch which worked for me.

```
diff --git a/py/builtinimport.c b/py/builtinimport.c
index 04ce6672..ae45d5e5 100644
--- a/py/builtinimport.c
+++ b/py/builtinimport.c
@@ -451,6 +451,9 @@ mp_obj_t mp_builtin___import__(size_t n_args, const mp_obj_t *args) {
                     // (the module that was just loaded) is not a package.  This will be caught
                     // on the next iteration because the file will not exist.
                 }
+
+                // only register the module if import was successful
+                mp_module_register(mod_name, module_obj);
             }
             if (outer_module_obj != MP_OBJ_NULL) {
                 qstr s = qstr_from_strn(mod_str + last, i - last);

diff --git a/py/objmodule.c b/py/objmodule.c
index f9363e37..bcce19eb 100644
--- a/py/objmodule.c
+++ b/py/objmodule.c
@@ -115,9 +115,6 @@ mp_obj_t mp_obj_new_module(qstr module_name) {
     // store __name__ entry in the module
     mp_obj_dict_store(MP_OBJ_FROM_PTR(o->globals), MP_OBJ_NEW_QSTR(MP_QSTR___name__), MP_OBJ_NEW_QSTR(module_name));
 
-    // store the new module into the slot in the global dict holding all modules
-    el->value = MP_OBJ_FROM_PTR(o);
-
     // return the new module
     return MP_OBJ_FROM_PTR(o);
 }

```
There might need to be some additional changes to other execution paths to avoid the call to `mp_module_register()` if it didn't actually load a module.

peterhinch, 2018-03-06:
I generally issue ctrl-d after failed loads. However some users have put this function into **boot.py**
```python
import sys
def reload(mod):
    mod_name = mod.__name__
    del sys.modules[mod_name]
    return __import__(mod_name)
```

dpgeorge, 2018-03-06:
Thanks for the report @tomlogic, but I think this is a duplicate of #967.  There are some comments in that issue about how it could be fixed.  It would be nice to fix this issue in a neat and compact way.

tomlogic, 2018-03-06:
Closing this issue out as a duplicate of #967.

closed 2018-03-06 by tomlogic