micropython/micropython #3658: Documentation for ESP32

bgamari, 2018-03-06:
It appears that the esp32 port has been merged (yay!) but there is no documentation for this port on [ReadTheDocs](http://docs.micropython.org/en/). Perhaps this should be fixed?

adritium, 2018-03-09:
You’re not gonna get documentation done by adding a ticket. I’ve tried it.

bgamari, 2018-03-09:
Ahh, quite right; I somehow thought there were already docstrings written and it were just a matter of flipping the switch on RtD. However, it seems a fair amount of effort remains to be done.

hansonap, 2018-11-27:
I plan on putting some time to this as I find it very irritating. In general I found that the micropython documentation to be lacking and I believe it is limiting it's potential. I've started to compile a list of differences. 

If you guys are also interested in helping, or already have resources that would help, it would be greatly appreciated.

mattytrentini, 2018-11-27:
That's timely; I've just started making a push for ESP32 documentation - in fact there are a couple of us that will be working on it at tomorrow night's [Melbourne MicroPython Meetup](https://melbournemicropythonmeetup.github.io/)!

If you give us a couple of days I should have a fork set up with a skeleton of the ESP32 docs to build on.

Also, if you have any specific part of the documentation you'd like to work on let us know here so we can focus on other areas. There is plenty to go around! :)

mattytrentini, 2018-12-02:
Started. If anyone has time to help that'd be great!

https://forum.micropython.org/viewtopic.php?f=2&t=5609