micropython/micropython #3666: WiPy/CC3200: usocket.socket.settimeout() not working for socket.getaddrinfo()

dmartauz, 2018-03-09:
In issue 2402 a solution was proposed to make socket.settimeout() working for socket.connect():
https://github.com/micropython/micropython/issues/2402
It seems to work well.

However it does not work for socket.getaddrinfo() as there seems to be fixed timeout of 19s.

```
import socket
import ssl
import time

def test(self):
	start = time.time()
	self.s = socket.socket(socket.AF_INET, socket.SOCK_STREAM, socket.IPPROTO_SEC)
	self.s = ssl.wrap_socket(self.s, cert_reqs=ssl.CERT_REQUIRED, ca_certs='/flash/cert/ca.pem')
	self.s.settimeout(3)
	try:
		print(socket.getaddrinfo("www.google.com", 443)[0][-1])
	except Exception as e:
		print(e)
	finish = time.time()
	print(finish - start)
	self.s.close()
```

dpgeorge, 2018-03-16:
> Anyway is there any way how to modify timeout of this method?

It seems to be a limitation of the CC3100 that getaddrinfo() is blocking and doesn't have a way to configure the timeout.  See eg here: https://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc31xx_cc32xx/f/968/p/532971/1945259

From that link there is a suggestion to reduce the number of retries, which defaults to the maximum of 32, to shorten the time that getaddrinfo() will spend waiting.

dmartauz, 2018-03-16:
For now my solution is to perform DNS lookup only once after boot and not with every socket.connect().

Anyway it might make sense to reduce number of retries (e.g. to 8). What is your opinion?