micropython/micropython #3667: esp8266: multi threading

misisnik, 2018-03-12:
Is it possible to build MicroPython for esp8266 with _thread support and how?

adritium, 2018-03-12:
This is not forum. For how-to, go to 
https://forum.micropython.org/viewforum.php?f=3&sid=3074c0fd0c4b1ec163589ef5d63accd6

dpgeorge, 2018-03-13:
> Is it possible to build MicroPython for esp8266 with _thread support and how?

No it's not possible because the esp8266 port doesn't implement multithreading at the lower levels (eg it doesn't implement switching of C stacks between two different threads).  This is a pretty difficult thing to add.

misisnik, 2018-03-13:
I found one Arduino core branch where it is possible https://github.com/bmellstrom/esp8266-arduino-threads
If I've understand how it works

mattytrentini, 2018-03-14:
@misisnik that's an example of _cooperative multitasking_. It's reliant on each 'thread' yielding control periodically so other threads can take over execution. It's quite different to the `_thread` model where there are _independent_ threads of execution running on the processor; as @dpgeorge mentioned, the ESP8266 only supports single-threaded execution so implementing multithreaded support is going to be difficult at best.

For a more _pythonic_ implementation of cooperative multitasking, take a look at Python's [asyncio](https://docs.python.org/3/library/asyncio-task.html) features. I would suggest starting by reading @peterhinch's excellent [Micropython asyncio tutorial](https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md) to learn about how asyncio works on Micropython. Many tasks that would typically be implemented with multithreading can now use asyncio instead and use less resources with simpler code.