micropython/micropython #3668: modify mp_arg_parse_all() to indicate if default value was assigned

adritium, 2018-03-13:
I suggest adding ability to determine whether the default value was assigned or whether the argument was passed in.

https://github.com/micropython/micropython/blob/bdc875e602f687bf0fb28c3a18565ffec4157f59/py/runtime.h#L54-L58

```
typedef _mp_arg_val_t {
    bool actual_arg;

    union value {
    bool u_bool;
    mp_int_t u_int;
    mp_obj_t u_obj;
    mp_rom_obj_t u_rom_obj;
    }
} mp_arg_val_t;
```

In https://github.com/micropython/micropython/blob/bdc875e602f687bf0fb28c3a18565ffec4157f59/py/argcheck.c#L74
everywhere there is an assignment to `out_vals`, there would be an additional line to set `bool actual_arg` like so:

```
out_vals[i].value = allowed[i].defval;
out_vals[i].actual_arg = false;
```

dpgeorge, 2018-03-15:
> you have an easy way of checking whether the default value was assigned or if the argument was passed in.

A good way to design a Python facing API is to allow the user to pass in the default value themselves.  Eg use `None` or `-1` as default values which indicate some default behaviour.  A good example of this is `str.split(sep=None, maxsplit=-1)`: this has 2 default values and both can be explicitly provided by the user.  This allows the user to provide a non-default value for `maxsplit` while still using the default `sep`, eg like `s.split(None, 3)`.  It's good to try and design your API like this as well.  Then you don't need special code to differentiate between the cases of "arg not supplied" and "default value supplied" because they become the same thing.

> This is necessary if there is no value that can be used as a sentinel because entire range is valid.

If you really need to distinguish "arg not supplied" from any other value then use a `MP_ARG_OBJ` type with a default value of `MP_OBJ_NULL`, which the user can never provide.

peterhinch, 2018-03-16:
The example `str.split(sep=None, maxsplit=-1)` works under CPython but throws an error under MicroPython, where it accepts only positional args:
```
MicroPython v1.9.3-362-g2ad555b on 2018-02-25; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> a = 'the quick brown fox jumps over the lazy dog'
>>> a.split(sep=None, maxsplit=-1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
```

dpgeorge, 2018-03-16:
> The example str.split(sep=None, maxsplit=-1) works under CPython but throws an error under MicroPython, where it accepts only positional args:

Yes you are right.  But it does work with just positional args, eg: `a.split(None, -1)` or `a.split(None, 2)`.

adritium, 2018-03-16:
@dpgeorge 

> If you really need to distinguish "arg not supplied" from any other value then use a MP_ARG_OBJ type with a default value of MP_OBJ_NULL, which the user can never provide.

But then I have to have my own additional code after calling `mp_arg_parse_all` to set the default values . . . which should be done by `mp_arg_parse_all`.

dhylands, 2018-03-16:
> But then I have to have my own additional code after calling mp_arg_parse_all to set the default values . . . which should be done by mp_arg_parse_all.

The default are provided at compile time. See this example:
https://github.com/micropython/micropython/blob/5edce4539b239eab9b045bb2fde18456f6fdbfe4/ports/stm32/timer.c#L524-L536
The .u_obj and .u_int assignments are setting the default values.

adritium, 2018-03-16:
@dhylands the point of this issue is:

> I suggest adding ability to determine whether the default value was assigned or whether the argument was passed in.

dhylands, 2018-03-16:
I guess Damien's answer of using .u_obj and assigning it to be MP_OBJ_NULL answered the question in my mind (this is a sentinel which you can't pass in from the python side). Can you explain why you need to know that the difference between "argument was provided with a default value" or "not provided and assigned a default value automatically"?

Using Damien's approach you wind up doing:
```C
mp_obj_t argVal = args[0].u_obj;
if (argVal == MP_OBJ_NULL) {
    argVal = SOME_DEFAULT; // You might not even need this line.
    // do whatever other special thing you need to do when this happens
}
```
Using your approach you wind up doing:
```C
mp_obj_t argVal = args[0].u_obj;
if (args[0].actual_arg) {
    // do whatever special thing you need to do when this happens
}
```
Your approach will waste considerably more flash memory, since almost all uses don't need to care about whether the argument was provided by the user or not.

Doing a quick grep on the stm32 port, adding a boolean to the  mp_arg_val_t struct would increase the flash size by about 496 bytes. I ran `grep MP_ARG_ *.c | wc -l` which counted 124 occurences in the ports/stm32 directory. Adding anything to that struct will use 4 bytes for the field, and 4 * 124 = 496.

adritium, 2018-03-16:
You make a good point.

closed 2018-03-16 by adritium