micropython/micropython #3675: Float type causes memory problem (not int)

Jomsanchez-gmv, 2018-03-20:
Using floats consumes memory without the GC releasing it.
In the following script, a float variable is incremented and the current state of the memory is shown and the free memory decreases.
```
import micropython

a = 1.1

while True:
	a = a + 1.1
	print(micropython.mem_info(1))
```
[image]

This same script made with a variable int and the sum of an int value. It does not cause these problems as seen in the following figure.
```
import micropython

a = 1

while True:
	a = a + 1
	print(micropython.mem_info(1))
```
[image]

This problem causes a MemoryException when the memory reaches zero, as seen in the following figure (a port with less available memory)
[image]

peterhinch, 2018-03-21:
I think your measurement technique is the issue here. Each pass of the loop will allocate RAM, however this will be released when a garbage collection (GC) is performed. To check for memory leaks you need to do a GC before the test starts, and again after it completes:
```python
import micropython, gc
gc.collect()
micropython.mem_info(1)
a = 1
while a < 1000000:
    a += 1.1
gc.collect()
micropython.mem_info(1)
```
Outcome: uses ~48 bytes.
(Takes ~ 12s on Pyboard V1.1)

Jomsanchez-gmv, 2018-03-22:
Thanks for your answer, I tested your script (unix port) and woks fine. My main problen was that I was using a port with MICROPY_ENABLE_IMMORTAL_GC option enabled and gc.collect() didn´t work.
I think that this "issue" is closed.

closed 2018-03-22 by Jomsanchez-gmv