micropython/micropython #3678: esp8266: webrepl_setup fails, no attribute 'hexilfy'

niallp, 2018-03-21:
Bringing up a fresh esp12 module with micropython, latest from git 

it doesn't appear to be setting up the filesystem initially and (likely as a consequence) I can't get webrepl_setup going.

It took a few tries to get the module flashed but seems to be consistent now, esp.check_fw() returns true.

On reset I get:
`Performing initial setup
Traceback (most recent call last):
  File "_boot.py", line 11, in <module>
  File "inisetup.py", line 38, in setup
  File "inisetup.py", line 8, in wifi
AttributeError: 'module' object has no attribute 'hexlify'
OSError: [Errno 1] EPERM
OSError: [Errno 1] EPERM

MicroPython v1.9.3-477-g7b0a020-dirty on 2018-03-21; ESP module with ESP8266
Type "help()" for more information.
>>> 
`
when I try to import webrepl_setup:
`>>> import webrepl_setup
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "webrepl_setup.py", line 102, in <module>
  File "webrepl_setup.py", line 71, in main
  File "webrepl_setup.py", line 40, in get_daemon_status
OSError: [Errno 1] EPERM
>>> 
`
Any ideas appreciated. I've erased flash and reload a couple times.

peterhinch, 2018-03-22:
You may get a quicker response to this in [the forum](https://forum.micropython.org/).

dpgeorge, 2018-03-27:
This looks the same as #3686: try doing a "make clean" first and then rebuild (also try rebuilding mpy-cross by doing "make clean" in the mpy-cross dir first).

niallp, 2018-03-28:
That was it, make clean was what it needed, thanks !

closed 2018-03-28 by niallp