micropython/micropython #3681: Add possibility dinamic freeze the modules

slavaza, 2018-03-22:
This means placing and dynamically linking the compiled modules during program execution, into program memory of microcontroller. It will allow save operation memory for variables and prevent superfluous often builds entire system. And do without a disk system at all.

adritium, 2018-03-22:
Not sure what you mean; give a complete example.

slavaza, 2018-03-22:
Now, it is possible to place the bytecode of the python in the program segment, by translating it into the C language representation, using the utility mpy-tool.py. However, this can only be done at the compilation and build stage of the project. While studying the project I did not find any possibility to place the bytecode into the Flash of the microcontroller at the runtime stage. Execute bytecode of  other modules is possible only in RAM. I believe that it would be very useful to freeze the modules in the process of executing the python programs.

adritium, 2018-03-22:
1. There is already a discussion here #2709; I've contributed to the discussion and I _think_ I know how to do this but I do not have a need for this at this moment. 
2. `persistentcode.c` provides the building blocks for implementing this

slavaza, 2018-03-22:
Thank You for hint. I will study it. I hope I can understand it myself.

adritium, 2018-03-22:
You should also close this issue and add any questions to #2709

slavaza, 2018-03-23:
@adritium  Good. I do not pretend to your fame as a co-author, or the glory of the author of the post # 2709. I just wanted to improve your interesting project. Despite the fact that I do not know English well, it seems to me that I have stated my thoughts correctly. And we are talking about some another possibility. I think you understood that. I will fulfill your demand after a while, with your permission.

slavaza, 2018-04-01:
I would have done it myself, if it were, it was necessary for someone other than me. It is possible to add another pool of reloadable frozen modules, by making a minor change into mp_find_frozen_mpy() function from "frozenmod.c" and redirecting the symbol table through pointer. The very functions of creating this pool can be implemented in a many variety of ways. However, this proposal was not arouse of interest, even in the issue of  [#2709](https://github.com/micropython/micropython/issues/2709) .

jonnor, 2024-07-02:
I believe this issue was covered in the linked issue.

dpgeorge, 2024-07-03:
Static freezing has existed for a long time.

Dynamic freezing is proposed by #8381 (and see also #12429 for an alternative).

closed 2024-07-03 by dpgeorge