micropython/micropython #3682: Is a port to STM32F070 possible?

askvictor, 2018-03-24:
I've been playing with the Lego Boost robot kit, and it uses an STM32F070RB microcontroller. It a port to this feasible? Getting it flashed easily would require using the BLE firmware update protocol; not sure how easy this would be to do.

tannewt, 2018-03-26:
16k RAM is pretty small for MicroPython. The micro:bit has an nRF51 with that much and basic functionality does work. So it is feasible.

We use the SAMD21 with a Cortex-M0+ so that won't be an issue.

I can't speak to the BLE firmware update.

dpgeorge, 2018-03-27:
I did recently get uPy compiled for and running on an STM32F091 with 256k flash and 32k RAM.  It runs quite well and can have quite a few features enabled.  So there's no problem supporting the STM32F0xx series of MCUs.

But the STM32F070 itself is pretty tight on flash, only 128k.  So uPy would run on it but it couldn't have an internal filesystem (only on external SPI flash if that was on the board) and it probably couldn't have floating point support enabled.

dpgeorge, 2018-05-28:
Commit e681372017d2af10ab167d2ab8bf4d005dd7cf69 added support for the NUCLEO-F091RC board, and commits just prior to that did all the work for supporting Cortex-M0 and STM32F0xx in the stm32 port.  Running on STM32F070 should now be straightforward.

closed 2018-05-28 by dpgeorge

laurensvalk, 2020-08-07:
@askvictor

> I've been playing with the Lego Boost robot kit, and it uses an STM32F070RB microcontroller. It a port to this feasible? Getting it flashed easily would require using the BLE firmware update protocol; not sure how easy this would be to do.

Fast forward two years, the answer is yes. You can try it live [here](https://code.pybricks.com/) and the source code is [here](https://github.com/pybricks/pybricks-micropython).