micropython/micropython #3684: Crashes with array('O') monkeybusiness

jepler, 2018-03-26:
I discovered that in CircuitPython, there are a number of crashes that can occur with 'O'-type arrays.  At least some of them apply to MicroPython, though I haven't triaged which ones.  Example:
~~~~
a = array.array('O')
a[:] = array.array('P', [4])
print(a)
~~~~
See adafruit/circuitpython#707 for more  and adafruit/circuitpython/#711 for my solutions.

I'll be happy to prepare a proper PR if the work in circuitpython looks like the right approach but doesn't apply cleanly via cherry-pick.  Please let me know if you'd like me to do that.

dpgeorge, 2018-03-27:
Thanks for the report @jepler.

The O, P and S typecodes are uPy extensions and mostly useful for the struct module, although they can also be used for array.  The bug reported above is actually "intended" behaviour: creating an array of objects and putting invalid pointers into it will lead to a crash.  These typecodes are powerful and if using them then you need to be careful.

dpgeorge, 2018-03-28:
@jepler I see that you have found quite a few other bugs with AFL, some of which are important and should be fixed.  Would you be interested in preparing a list of all the bugs and then we can decide which are important?