micropython/micropython #3686: websocket for ESP32

plugowski, 2018-03-26:
I compiled firmware using newest micropython library, but websocket seems to be not implemented at all:

```
>>> import websocket
>>> dir(websocket)
['__class__', '__name__', 'wake_reason']
```

dpgeorge, 2018-03-27:
It looks like your build got corrupted (websocket shouldn't have wake_reason).  Please just try "make clean" and then rebuild.

plugowski, 2018-03-27:
I tried couple times, finally I went back couple revisions back and then it works again.
I've got similar problem with ssd1306 driver, it shout that cant convert bytearray to int, and after back with revisions it started to work...

dpgeorge, 2018-03-27:
> finally I went back couple revisions back and then it works again.

Which revision in particular is working?

plugowski, 2018-03-27:
This one: 0d5bccad

dpgeorge, 2018-03-28:
I tried building the latest code for the esp32 (at cf1d6df05acc22b641fb97de6a87b100b4fcfcb1) and it works just fine for me (websocket module shows the right contents).

@plugowski please can you specify all your build settings: host OS, xtensa toolchain, ESP IDF commit, MicroPython version, etc.  (Did you update to the latest xtensa esp32 toolchain?)

plugowski, 2018-03-28:
Hi,

Okay I download all tools from the scratch and **it works now** for me.

```
macOS High Sierra 10.13.3 (17D102)
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
esp-idf commit: 3ede9f011b50999b0560683f9419538c066dd09e
micropython commit: 4d409b8e3276c2e40191ada9726780a0dcc776ba
```

So issue can be closed :)

closed 2018-03-28 by plugowski