micropython/micropython #3687: Not cacert parameter in micropython/extmod/modussl_mbedtls.c

Sunrise17, 2018-03-29:
Dear @dpgeorge ,
I am trying to establish SSL connection to AWS IoT via ESP32. Normally i have below three files (certs and key) in flash folder. 
KEY_PATH = "/flash/cert/deviceCert.key"
CERT_PATH = "/flash/cert/deviceCertAndCACert.pem"
CACERT_PATH = "/flash/cert/root.pem"

In first, i defined the ssl_params={"certfile":CERT_PATH, "keyfile":KEY_PATH, "ca_certs":CACERT_PATH}) as like that but i was always getting "Error in mqtt connect: [Exception] TypeError: extra keyword arguments given." error. 

Then, i have checked the modussl_mbedtls.c in extmod and SSL parameters are defined as like in struct below (Line 59).
    struct ssl_args {
        mp_arg_val_t key;
        mp_arg_val_t cert;
        mp_arg_val_t server_side;
        mp_arg_val_t server_hostname;
    };

i have adjusted the ssl_params={"cert":CERT_PATH, "key":KEY_PATH})
Try to do that without using cacert.

Then, it started to try establishing connection but i have received following error then board reboots.

    assertion "ret == 0" failed: file "../../extmod/modussl_mbedtls.c", line 182, function: socket_new
    abort() was called at PC 0x40135e2f on core 0

    Backtrace: 0x4008c983:0x3ffb61f0 0x4008c9af:0x3ffb6210

    Rebooting...
    0�?anic'ed ( 0�?anic'ed ( 0�?anic'ed ( 0�?anic'ed ( 0�?anic'ed ( 0�?anic'ed ( 0�?anic'ed (ets Jun  8 2016 00:22:57

    rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

I will be glad if it is possible to add "cacert" parameter inside arguments of ussl module.

Thanks for your efforts...

Sunrise17, 2018-03-29:
I have applied based on your guidance in the [link](https://github.com/micropython/micropython-esp32/issues/132) Directly i added key and cert inside of main.py as follows;

    # this is the mbedtls test RSA key; you must change it to your own for production!
    key = """-----BEGIN RSA PRIVATE KEY-----
    MIIEpAIBAAKCAQEAwU2j3efNHdEE10lyuJmsDnjkOjxKzzoTFtBa5M2jAIin7h5r
    lqdStJDvLXJ6PiSa/LY0rCT1d+AmZIycsCh9odrqjObJHJa8/sEEUrM21KP64bF2
    2JDBYbRmUjaiJlOqq3ReB30Zgtsq2B+g2Q0cLUlm91slc0boC4pPaQy1AJDh2oIQ
    Zn2uVCuLZXmRoeJhw81ASQjuaAzxi4bSRr/QuKoRAx5/VqgaHkQYDw+Fi9qLRF7i
    GMZiL8dmjfpd2H3zJ4kpAcWQDj8n8TDISg7v1t7HxydrxwU9esQCPJodPg/oNJhb
    y3NLUpbYEaIsgIhpOVrTD7DeWS8Rx/fqEgEwlwIDAQABAoI......
    -----END RSA PRIVATE KEY-----
    """

    # this is the mbedtls test certificate; you must change it to your own for production!
    cert = """-----BEGIN CERTIFICATE-----
    MIIDNzCCAh+gAwIBAgIBAjANBgkqhkiG9w0BAQUFADA7MQswCQYDVQQGEwJOTDER
    MA8GA1UEChMIUG9sYXJTU0wxGTAXBgNVBAMTEFBvbGFyU1NMIFRlc3QgQ0EwHhcN
    MTEwMjEyMTQ0NDA2WhcNMjEwMjEyMTQ0NDA2WjA0MQswCQYDVQQGEwJOTDERMA8G
    A1UEChMIUG9sYXJTU0wxEjAQBgNVBAMTCWxvY2FsaG9zdDCCASIwDQYJKoZIhvcN
    AQEBBQADggEPADCCAQoCggEBAMFNo93nzR3RBNdJcriZrA545Do8Ss86ExbQWuTN
    owCIp+4ea5anUrSQ7y1yej4kmvy2NKwk9XfgJmSMnLAofaHa6ozmyRyWvP7BBFKz
    NtSj+uGxdtiQwWG0ZlI2oiZTqqt0Xgd9GYLbKtgfoNkNHC1JZvdbJXNG6AuKT2kM.....
    -----END CERTIFICATE-----
    """

    client = MQTTClient(client_id=MQTT_CLIENT_ID, server=MQTT_HOST, port=MQTT_PORT, keepalive=10000, ssl=True, ssl_params={ "key":key, "cert":cert, "server_side":True })

mqtt file:
        if self.ssl:
            import ssl
            self.sock = ssl.wrap_socket(self.sock, **self.ssl_params)

 However, right now, It is waiting during the connection and after 15 secs, i have received following error

    mbedtls_ssl_handshake error: -7280
    Traceback (most recent call last):
      File "main.py", line 300, in <module>
      File "main.py", line 258, in mqtt_connect
      File "mqtt.py", line 69, in connect
    OSError: [Errno 5] EIO

Sunrise17, 2018-04-10:
Dear @dpgeorge ,
Is it not possible to add following module (mosussl.c in cc3200) below, into ESP32 ports?
[https://github.com/micropython/micropython/blob/master/ports/cc3200/mods/modussl.c](url)

silps, 2018-06-30:
@Sunrise17 Did you manage to get it working ? I am also trying to get SSL working for my WebSocket connection on the ESP32.

Sunrise17, 2018-06-30:
@silps , Unfortunately over mqtt to AWS IoT, i could not succeed to connect but normally it's connecting well other ssl required socket servers (without ca_cert parameter).

silps, 2018-06-30:
@Sunrise17 Can you give me some example code and possibly a server hostname (if public) to try it on? I tried something like this.
```
>>> import socket
>>> s = socket.socket()
>>> import ssl
>>> ssl.wrap_socket(s)
mbedtls_ssl_handshake error: -80
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
```

Sunrise17, 2018-07-01:
@silps , You can use following code below;

    KEY_PATH = "/flash/cert/client.key"
    CERT_PATH = "/flash/cert/client.crt"
    HOST, PORT = "45.255.66.124", 5100

    with open(KEY_PATH, 'rb') as f:
        key1 = f.read()

    with open(CERT_PATH, 'rb') as f:
        cert1 = f.read()

    s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)
    addr = usocket.getaddrinfo(HOST, PORT)[0][-1]
    s.connect(addr)
    sock = ussl.wrap_socket(s, key = key1, cert = cert1)
    sendObject = 'hello'
    bytesObject = sendObject.encode()
    sock.write(bytesObject)
    recvData = sock.read(4096).decode()

silps, 2018-07-01:
@Sunrise17 I tried using the key and cert above you provided and connect to google.com. Once I use the wrap_socket my ESP32 reboots.
```
>>> s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)
>>> addr = usocket.getaddrinfo(HOST, PORT)[0][-1]
>>> addr
('216.58.212.46', 443)
>>> s.connect(addr)
>>> sock = ussl.wrap_socket(s, key=key, cert=cert)
assertion "ret == 0" failed: file "../../extmod/modussl_mbedtls.c", line 176, function: socket_new
abort() was called at PC 0x4013893f on core 0
```

Sunrise17, 2018-07-01:
@silps what is the version of your micropython on ESP32 board? if it is not 1.9.4, please flash your device with following fw: [esp32-20180701-v1.9.4-227-gab02abe9.bin](http://micropython.org/resources/firmware/esp32-20180701-v1.9.4-227-gab02abe9.bin)

silps, 2018-07-01:
@Sunrise17 I am using the latest firmware yes, just updated it also and tried again.
```
MicroPython v1.9.4-227-gab02abe9 on 2018-07-01; ESP32 module with ESP32
Type "help()" for more information.
```

Sunrise17, 2018-07-01:
It is connecting to my SSL server as follows but i could not connect your IP, i think you should check your configuration. In addition, please be aware that your ESP board is connected to internet, because in this last version, it delays and sometimes it connects after 15secs, that's why you should add some delay before making your ssl request.

    Request_Sent::
    HTTP/1.1 200 OK
    Connection: close
    Content-Type: text/x-python
    Last-Modified: Wed, 27 Jun 2018 20:32:48 GMT
    Content-Length: 367

silps, 2018-07-01:
Ok thanks a lot for the help. I will try to figure it out. Seems like some configuration issue yes, with the cert maybe etc.

fatihak, 2018-09-14:
Hey @Sunrise17, have you found a way to fix the CA_cert parameter in in ssl_args? I am also trying to connect to the AWS iOT with ESP32 and micropython.

dpgeorge, 2022-07-12:
I'll close this as a duplicate of #3646, with the issue being that `ca_certs` and `cert_reqs` arguments are not implemented yet.

closed 2022-07-12 by dpgeorge