micropython/micropython #3692: overwrite builtin object __new__ method cased a segment fault

learnforpractice, 2018-03-31:
here is the code that works like calm in cpython but caused a panic in the recent built micropython:

```python
class List(list):
    def new(cls, value):
        return super(List, cls).new(cls, value)

l = List([1,2,3])
```

peterhinch, 2018-03-31:
Subclassing built-in types is not fully supported in MicroPython. See [this doc](https://github.com/micropython/micropython/wiki/Differences).

learnforpractice, 2018-04-02:
Yes, I was wondering why It doesn't throw a exception instead of crash the process, I will refer to the code to find a clue.

learnforpractice, 2018-04-03:
Can't not reproduce, will catch it up later.

closed 2018-04-03 by learnforpractice

linguistgeek, 2018-09-11:
I'm running into super().__new__(cls) recursions that cause segfaults depending on the architecture that I compile for. This is in the unix port. My test script:
```
class Test(type):
    def __new__(cls):
    cls.testing = []
    base_stuff = list(name for name in dir(cls))
    for item in base_stuff:
        cls.testing.append(item)
    return super().__new__(cls)

class Test2(Test):
    def __str__(self):
        return self.__module__.split('/', 2).pop()
print("About to call super now")
garr = Test2()
print("Passed the test if no segfault")
```
Interestingly enough, on the architectures this doesn't segfault on, there's some crazy recursion going on if you compile with `-DMICROPY_DEBUG_VERBOSE=1` . It's like it's iterating through things that aren't in the dict as well as some that are. Example from x86 compiled that did finally work with no segfault:

```
Looking through locals_dict
mp_obj_class_lookup: Looking up __str__ in Test2
mp_obj_class_lookup: Looking up __sub__ in Test2
mp_obj_class_lookup: Looking up __sub__ in Test
mp_obj_class_lookup: Looking up __sub__ in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up __traceback__ in Test2
mp_obj_class_lookup: Looking up __traceback__ in Test
mp_obj_class_lookup: Looking up __traceback__ in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up __truediv__ in Test2
mp_obj_class_lookup: Looking up __truediv__ in Test
mp_obj_class_lookup: Looking up __truediv__ in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up __xor__ in Test2
mp_obj_class_lookup: Looking up __xor__ in Test
mp_obj_class_lookup: Looking up __xor__ in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up * in Test2
mp_obj_class_lookup: Looking up * in Test
mp_obj_class_lookup: Looking up * in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up _ in Test2
mp_obj_class_lookup: Looking up _ in Test
mp_obj_class_lookup: Looking up _ in type
mp_obj_class_lookup: No more parents
mp_obj_class_lookup: Looking up %#o in Test2
mp_obj_class_lookup: Looking up %#o in Test
mp_obj_class_lookup: Looking up %#o in type
mp_obj_class_lookup: No more parents
```
Eventually the above iterated through random parts of frozen modules, errors, and datatypes before finally returning without segfault. But in other situations, segfault. Maybe the segfault is just from too much recursion? I know stack size seems to be dependent on the dword size for the architecture, so it would possibly make sense to segfault in some places and not others depending on what it was compiled for.