micropython/micropython #3699: Python for the 8bit AVR mocrocontroller

slavaza, 2018-04-03:
I has compiled Micropython in minimal configuration, for the AVR8 microcontrollers. Just for the sake of curiosity. The footprint is shown below.

# AVR uPython LLVM with Compiller
# text    data     bss     dec     hex    filename
# 96580   11430    2288  110298   1aeda build/firmware.elf

Maybe it will be interesting to someone?

3bb, 2018-04-04:
Yes :+1:

slavaza, 2018-04-10:
@rolandvs Good. Unfortunately I can not demonstrate a workable port to AVR8. This would require significant costs as for me. You can install and try compile by yourself by installing the `avr8-dummy` port from my repository.

aofek, 2021-06-23:
very good idea
it is about time to use a single language on embedded systems
8b ,16b , 32b , 64b they all should use one language, preferably python

learn st8, st32,esp32,msp430, Arm Mbed, Arduino, Adafruit and many more....  one spend more time on learning language than on applying algorithms....

Thanks Slavaza

jonnor, 2023-07-30:
For reference, the mentioned repository appears to be https://github.com/slavaza/micropython-avr8

mattytrentini, 2023-07-31:
It's good to see a port of MicroPython to AVR, even if it's fairly rudimentary. However, the core team is focussed on 32-bit microcontrollers and it seems low-value to target 8-bit devices, especially since, for that class of device, only a small percentage has sufficient flash. 

We will definitely help answer questions and provide guidance about the porting effort. But I think it unlikely that such a port will be accepted upstream unless someone will show clear willingness to own and, especially, support such a port. 

So, I think it reasonable that we close this ticket, at least until such a person steps forward.

closed 2023-07-31 by mattytrentini