micropython/micropython #3706: H7 I2C4 missing

rolandvs, 2018-04-09:
When I added the I2C4 port definition to mpconfigport.h I came across some missing things:

1) HAL is missing some labels pertaining the I2C4 bus
2) the I2C.c is missing creation in when making a new object
3) the I2C.h is missing the I2C4 mention.

Added to `mpconfigport.h` of `boards/NUCLEO_H743ZI`:
```
#define MICROPY_HW_I2C4_SCL         (pin_F14)
#define MICROPY_HW_I2C4_SDA         (pin_F15)
```

Added to `stm32/i2c.h`

```
extern I2C_HandleTypeDef I2CHandle4;
```

Added to `stm32/i2c.c` at `make_new`
```
        #ifdef MICROPY_HW_I2C4_NAME
      } else if (strcmp(port, MICROPY_HW_I2C4_NAME) == 0) {
            i2c_id = 4;
        #endif

```

And had to add to `STM32H7xx_HAL_Driver/inc/Legacy/stm32_hal_legacy.h`
```
#define __I2C4_CLK_ENABLE __HAL_RCC_I2C4_CLK_ENABLE
#define __I2C4_CLK_SLEEP_DISABLE __HAL_RCC_I2C4_CLK_SLEEP_DISABLE
#define __I2C4_CLK_SLEEP_ENABLE __HAL_RCC_I2C4_CLK_SLEEP_ENABLE
#define __I2C4_FORCE_RESET __HAL_RCC_I2C4_FORCE_RESET
#define __I2C4_RELEASE_RESET __HAL_RCC_I2C4_RELEASE_RESET
```
For it to compile without errors...

dpgeorge, 2018-04-10:
> And had to add to STM32H7xx_HAL_Driver/inc/Legacy/stm32_hal_legacy.h ...

In 69bf23c9cfacae4745c260b9a75ca5144ffb353b I fixed stm32/i2c.c so that it can use I2C4 with the new `__HAL_RCC_xxx` macros, so there shouldn't be a need to patch the HAL itself.

> Added to stm32/i2c.h ...
> Added to stm32/i2c.c at make_new ...

Thanks for the pointers, I fixed these in 22f1414abb538e80bda6c9c7123534923558d0fb

> Added to mpconfigport.h of boards/NUCLEO_H743ZI:

I didn't enabled I2C4 for this board.  Feel free to submit a PR for it if you like :)

rolandvs, 2018-04-10:
@dpgeorge I will submit a PR later on. Thanks!

closed 2018-04-10 by rolandvs