micropython/micropython #3710: Update documenation of machine.RTC

nagylzs, 2018-04-10:
The documentation has incorrect signature. There is no RTC.init method. The RTC.datetime method has a different signature: it has no tzinfo parameter, and the "day of week" parameter follows the day parameter (instead of the hour).

For details see: https://stackoverflow.com/questions/49757733/micropython-on-esp8266-rtc-does-not-have-an-init-method/49758737#49758737

schinckel, 2018-04-13:
Likewise, the ``RTC.irq()`` method appears to not take a ``handler`` argument.

schinckel, 2018-04-13:
I believe this is just the `esp8622` port that exhibits this behaviour.

dpgeorge, 2019-12-17:
In 04e7aa0 the docs were updated to mention which RTC methods are missing on esp8266.

closed 2019-12-17 by dpgeorge