micropython/micropython #3714: How to implement cross support in socket Access-Control-Allow-Origin

anjamago, 2018-04-11:
I have a program created in my esp8266 which works well,
I have created a web application on my pc which I connect to my ESP network
my web is commented by http requests but this returns me an error because the origin does not have support for corss Access-Control-Allow-Origin
How can I implement, enable support for cors from my esp ???

belyalov, 2018-04-15:
Hi,

If I understood you right you're trying to build some Rest API using `AJAX`.
So it is probably better to use existing web-server framework like [tinyweb](https://github.com/belyalov/tinyweb) or [picoweb](https://github.com/pfalcon/picoweb).

anjamago, 2018-04-16:
Hello,
if exactly what little thing is to connect with ajax I see those two possible ways to solve my problem but I do not know how to implement it in the esp8266 since I'm just entering this world of iot and I do not know how to implement new packages in this module

anjamago, 2018-04-17:
Hello,
if I have already looked at the examples and I have the idea of ​​how to implement it but what I need to know on time is how to install the tinyweb package
in micropython try with upip but this message is what answers me I have no knowledge of how to install the package

upip.install("tinyweb")
Installing to: /lib/
Error installing 'tinyweb': b'302', packages may be partially installed

belyalov, 2018-04-17:
@anjamago actually it is not a good idea to use `upip` - just because you're running your app on `esp8266` which has really few memory (about 32k).

So even if you be able to download module it probably fail to compile (due to out of memory). So the better way is use **frozen** modules which stands for "precompiled" modules during firmware compilation. It looks complicated (especially if you're new to micropython) however it is probably only one way to do so.

If you aren't feeling yourself comfortable with building firmware - you can start with firmware from **tinyweb** project - simply go to [tinyweb releases](https://github.com/belyalov/tinyweb/releases) and flash latest version.

Then just do (no need to install it - it is already frozen inside firmware)
```
import tinyweb
```

Neon22, 2018-04-18:
@anjamago - Konstantin's response is how you should do it.
    Specifically:
       - instead of using micropython binary use the one on this page:
           - https://github.com/belyalov/tinyweb
      - install it using the instructions in the section labelled -
      Quickstart - just like you would install micropython
    This micropython binary already contains tinyweb. This means you
      don't need upip. As explained, if you use upip it won't work.
      There is not enough memory on the ESP8266 for upip to work.
    Good luck.
    
    
    On 4/18/2018 7:00 AM, anjamago wrote:
    
    
      Hello,
        if I have already looked at the examples and I have the idea of
        ​​how to implement it but what I need to know on time is how to
        install the tinyweb package
        in micropython try with upip but this message is what answers me
        I have no knowledge of how to install the package
      upip.install("tinyweb")
        Installing to: /lib/
        Error installing 'tinyweb': b'302', packages may be partially
        installed
      —
        thread.

anjamago, 2018-04-24:
Hello I was implementing it with the firmware as it is in tinyweb but it still tells me that the module does not exist

anjamago, 2018-04-25:
@belyalov  I installed the following firmware [firmware_esp8266.bin](https://github.com/belyalov/tinyweb/releases)

[esp module](https://i0.wp.com/electronilab.co/wp-content/uploads/2016/02/NodeMCU-%E2%80%93-Board-de-desarrollo-con-m%C3%B3dulo-ESP8266-WiFi-y-Lua-1.jpg?fit=1200%2C900&ssl=1)

**sudo esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash**
`
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 9.1s
Hard resetting via RTS pin...

`
**sudo esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fm dio 0 firmware_esp8266.bin** 

`
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 594336 bytes to 388495...
Wrote 594336 bytes (388495 compressed) at 0x00000000 in 34.4 seconds (effective 138.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`
I was able to implement the tinyweb module but I do not know why the previous attempts to install the firmware did not load the module

belyalov, 2018-04-25:
@anjamago so as I understood you correctly - now you've got working setup? 😀

anjamago, 2018-04-25:
@belyalov it is correct I have the functional configuration I will do tests with the program that I have created and the implementation of tinyweb I hope it works for me without any problem

belyalov, 2018-04-25:
@anjamago cool! Congrats! :)

Feel free to open issue on `tinyweb` if any.. 👍

dpgeorge, 2018-05-17:
It sounds like this is more a discussion for the forum or other projects and not an issue with MicroPython itself, so closing the issue.

closed 2018-05-17 by dpgeorge