micropython/micropython #3715: Do we want to add peripherals to the board files?

rolandvs, 2018-04-12:
Just while upgrading my collection of STM32 boards I was looking through the pin definitions and one can debate if they are complete. Other boards also suffer from incompleteness. So now the question pops up to support only the bare micropython or add the peripherals too.

For the `STM32F4DISC` board the following 'needs' to be added. Also like with the `OTG_FS_POWER_SWITCH` it seems reasonable to add a disable at startup time (like what is done with the `NUCLEO_H743ZI` board implementation.

There is however a driver for the `LIS302DL` in python. It would be nice to include a driver for the `MP45DT02` microphone and `CS43L22` 24-bit audio DAC.

So for starters this needs to be added to the `pins.csv`:
```
SW,PA0
LED_GREEN,PD12
LED_ORANGE,PD13
LED_RED,PD14
LED_BLUE,PD15
VBUS_FS,PA9
OTG_FS_ID,PA10
OTG_FS_DM,PA11
OTG_FS_DP,PA12
OTG_FS_POWERSWITCH,PC0
OTG_FS_OVERCURRENT,PD5
LIS302DL_CE,PE3
LIS302DL_INT2,PE1
LIS302DL_SCK,PA5
LIS302DL_SDO,PA6
LIS302DL_SDI,PA7
MP45DT02_CLK,PB10
MP45DT02_DOUT,PC3
CS43L22_MCLK,PC7
CS43L22_LRCK,PA4
CS43L22_SCLK,PC10
CS43L22_SDIN,PC12
CS43L22_RESET,PD4
CS43L22_SCL,PB8
CS43L22_SDA,PB9
BOOT1,PB2
SWDIO,PA13
SWCLK,PA14
SWO,PB3
```

First would be to add these peripheral definitions for all the DISCO and NUCLEO boards. Do we like this approach?

dpgeorge, 2018-05-02:
> I was looking through the pin definitions and one can debate if they are complete. Other boards also suffer from incompleteness. So now the question pops up to support only the bare micropython or add the peripherals too.

The main reason not all the pin definitions are in any given pins.csv is simply because no one got round to adding and testing them.  Feel free to add them and submit PRs for boards that you have.

For the naming scheme of the pins: if, eg, PB8 is only connected to CS43L22_SCL then it makes sense to name it this way.

schef, 2021-05-04:
Just to clarify things..
What needs to be done to be able to use the missing pins?
I have STM32F746ZG and am missing a lot of pins.
Thanks

dpgeorge, 2021-05-05:
> What needs to be done to be able to use the missing pins?

You need to add the pins to the `pins.csv` file for your board.

If you don't want to give them special names and just refer to them with their MCU name, then adding this would be enough:
```
,PA0
,PA1
,PA2
,PA3
,PA4
,PA5
,PA6
,PA7
,PA8
,PA9
,PA10
,PA11
,PA12
,PA13
,PA14
,PA15
,PB0
,PB1
,PB2
,PB3
,PB4
,PB5
,PB6
,PB7
,PB8
,PB9
,PB10
,PB11
,PB12
,PB13
,PB14
,PB15
,PC0
,PC1
,PC2
,PC3
,PC4
,PC5
,PC6
,PC7
,PC8
,PC9
,PC10
,PC11
,PC12
,PC13
,PC14
,PC15
```

schef, 2021-05-05:
It works. Thanks!