micropython/micropython #3716: the stack SIZE is treated as a root pointer

adritium, 2018-04-12:
https://github.com/micropython/micropython/blob/d12483d93662d0aa1a421f4f8348c7e960c25183/py/mpstate.h#L219-L237

This can't be right.

`stack_limit` is the SIZE of the stack.

According to the logic in `gc_collect_root`, if the stack SIZE passes the checks in `VERIFY_PTR()` - which are a function of user inputs and some contstants - the gc will treat the stack SIZE as a root pointer.

dpgeorge, 2018-04-13:
Since the GC is conservative it doesn't matter if some values are considered pointers when in fact they are not (but it must be that anything which is a pointer is never missed).  So it's OK that `stack_limit` is in the root pointer section.  There are other pure numbers that are also there, like `sched_state` and `sched_sp`.

Of course this is not ideal, it would be better if the root pointer section only contained true pointers.  That would help eliminate any stale garbage.

Since the nlr code is now all implemented in .c files (it was in .S originally) it's now possible to consolidate the entries in `mp_state_thread_t` to put the root pointers at the end of that struct and to not scan entries in that struct that are not pointers.  But it requires some careful work to do that, and lots of testing.

closed 2018-04-13 by adritium

dpgeorge, 2018-05-13:
> it's now possible to consolidate the entries in mp_state_thread_t to put the root pointers at the end of that struct and to not scan entries in that struct that are not pointers

This was done in 749b16174b5292b2680ab027a6a1b3874e22ea43