micropython/micropython #3722: porting micropython on nucleo -429ZI

Abdelrahman-Elnaggar, 2018-04-15:
can any one help me  port micropython on nucleo stm32 429ZI as i can't find firmware for this board , i would really appreciate the help.

rolandvs, 2018-04-15:
Go to the `stm32` directory and type: `make BOARD=NUCLEO_F429ZI`

Abdelrahman-Elnaggar, 2018-04-17:
Thank you rolandvs so much for your help , i used your command and i found the firmware but when i try to upload the dfu on the board  using this command:
dfu-util -a 0 -d 0483:df11 -D build-NUCLEO_F429ZI/firmware.dfu
it doesn't work as it gives me :
dfu-util : No DFU capable USB device available
and i put the board on a boot-loader mode and the command didn't work.
I am running Linux as a virtual machine so i don't know if the problem relate to the virtual machine by any means which i highly doubt that it doesn't. i would really appreciate the help.

[image]

rolandvs, 2018-04-17:
@Abdelrahman-Elnaggar Are you sure it is in dfu mode? The connection is between BT0 and VDD (3V) and then hit reset or power cycle. Sometimes it is required to do the power cycle instead of hitting the reset button. So you could check that. Hope that helps.

Abdelrahman-Elnaggar, 2018-04-17:
thank you rolandvs and chrismas9 very much , i really appreciate your help. i managed to install dfu on windows using dfuse  as i was using the stlink debug port so by changing to the user USB port as chrismas9 said  everything worked fine.

closed 2018-04-17 by Abdelrahman-Elnaggar