micropython/micropython #3723: bug - gc.collect() kills items stored with mp_store_name, mp_store_global, mp_module_register

adritium, 2018-04-16:
In `mp_init()`, both **locals** and **globals** are set to the same dictionary `dict_main`
https://github.com/micropython/micropython/blob/d12483d93662d0aa1a421f4f8348c7e960c25183/py/runtime.c#L95-L100

and `dict_main` is part of the root pointer section

https://github.com/micropython/micropython/blob/d12483d93662d0aa1a421f4f8348c7e960c25183/py/mpstate.h#L153-L189

as is `mp_obj_dict_t mp_loaded_modules_dict;` which is where the `mp_module_register` dictionary stores its items.

Then, I add my module using any of the 3 methods

`mp_store_global(MP_QSTR_device, (mp_obj_t)&device_module);`
or
`mp_store_name(MP_QSTR_device, (mp_obj_t)&device_module);`
or
`mp_module_register(MP_QSTR_device, (mp_obj_t)&device_module);`

In uPy, I execute the following:
```
dir(device) # works correctly
gc.collect()
dir(device) # I get a C-code memory access trap
```

My hierarchy is:
```
device.top.
    item1
    item2
```

And everything below `device` is **_dynamically created_** and assigned to the dictionary of its parent.
I think this is the critical detail. Now that I think about it, I've not seen an example where a module is completely created on the fly.

Because the garbage collector is not documented, I only have the forum to go by which says
 
_as long as there is a chain of pointers starting from the root, the gc will not collect/delete an object_

I believe I've satisfied the _chain of pointers_ requirement via the dictionaries. If I do not `gc.collect()`, everything works correctly so I know the pointers are linked up.

Obviously, I'm not getting the full picture regarding the garbage collector.

dhylands, 2018-04-16:
If the device (module object) is not allocated dynamically (you've allocated it as a global variable) and the device object isn't included in the root pointers, then the gc won't know that it contains pointers into the heap.

The solution in this case would be to allocate the module object from the heap and populate it. And this is how modules created in python work.

adritium, 2018-04-16:
Yes. Oops. I fixed it by putting the `device_module` in the root pointers section.

closed 2018-04-16 by adritium