micropython/micropython #3725: Undefined symbols in esp32 link

readysteadygo2006, 2018-04-17:
Hi,

I'm trying to build the micropython for esp32.
I'm using the tool chain version as of: 16/4/18:
```
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/pgrant/esp32/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf

Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 
```

and the supported sha1 within IDF sdk: 3ede9f011b50999b0560683f9419538c066dd09e

Build fails with following error for undefined symbols:

```
LINK build/application.elf
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_api.o):(.iram1+0x8): undefined reference to `coexist_printf'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_api.o): In function `coex_wifi_request':
(.iram1+0x1f): undefined reference to `coexist_printf'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_api.o): In function `coex_wifi_request':
(.iram1+0x4f): undefined reference to `coexist_printf'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_api.o): In function `coex_wifi_release':
(.iram1+0x7b): undefined reference to `coexist_printf'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_api.o): In function `coex_bt_request':
(.iram1+0xab): undefined reference to `coexist_printf'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libcoexist.a(coexist_core.o):(.iram1+0xd0): more undefined references to `coexist_printf' follow
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_rf_phy_enable+0xc): undefined reference to `esp_modem_sleep_register'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_rf_phy_enable+0x10): undefined reference to `esp_modem_sleep_exit'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_rf_phy_enable':
(.text.wifi_rf_phy_enable+0x68): undefined reference to `esp_modem_sleep_register'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_rf_phy_enable':
(.text.wifi_rf_phy_enable+0x70): undefined reference to `esp_modem_sleep_exit'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_rf_phy_disable+0x4): undefined reference to `esp_modem_sleep_enter'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_rf_phy_disable+0x8): undefined reference to `esp_modem_sleep_deregister'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_rf_phy_disable':
(.text.wifi_rf_phy_disable+0x55): undefined reference to `esp_modem_sleep_enter'
/home/pgrant/esp32/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_rf_phy_disable':
(.text.wifi_rf_phy_disable+0x61): undefined reference to `esp_modem_sleep_deregister'
Makefile:656: recipe for target 'build/application.elf' failed
make: *** [build/application.elf] Error ****

```
Can anyone help!

Best Regards

dpgeorge, 2018-04-20:
I'm not sure what the problem would be... I can only suggest to do a `make clean` first, then rebuild.  Also be sure to do `git submodule update --init` in the ESP IDF directory.

applio, 2018-04-24:
In case the original poster does not return with an update, I can confirm that having encountered this same build failure and accompanying errors, performing the suggested `git submodule update --init` did solve my problem and my next attempted build of micropython succeeded.

At least in my case, my steps leading up to this mistake were:
1. Checked out the latest ESP-IDF (and its submodules) from its git repo
2. Attempted to build micropython which failed but also very helpfully warned me that I was not using a supported release of ESP-IDF at the head of the build output
3. Performed a git checkout of the supported version of ESP-IDF
4. Did a `make clean` in the micropython source directory
5. Attempted to build micropython again and encountered the issue originally described here

Going back to ESP-IDF and updating its submodules then returning to build micropython worked.

dpgeorge, 2018-05-18:
> performing the suggested git submodule update --init did solve my problem and my next attempted build of micropython succeeded.

This step is documented in the README.

closed 2018-05-18 by dpgeorge