micropython/micropython #3728: Delete key broken in REPL

Sembazuru, 2018-04-17:
I discovered an issue with the REPL, originally reported to the Mu repository as [issue #225](https://github.com/mu-editor/mu/issues/225#issuecomment-381986417). It was suggested that I also report this issue here.

Here is what I said there:

> I have Mu configured to start in Adafruit CircuitPython mode, I really haven't used any other modes.
>
> In the code editor the DELETE and BACKSPACE keys function as expected.
>
> I then click the REPL button to open the REPL/Serial pane. I then press any key to "enter the REPL". I execute a command (for example help("modules")). Then I press up arrow to get the previous command and use the left arrow to move the cursor into the previous command. This is where both DELETE and BACKSPACE both function like BACKSPACE.
>
> If the REPL is running internally on the circuit python device, the issue might actually be with the device firmware instead of Mu. Or, maybe the wrong keycode is being sent for the DELETE key? Here is the version string for the installed circuit python on my device:
> Adafruit CircuitPython 2.2.4 on 2018-03-07; Adafruit Gemma M0 with samd21e18

peterhinch, 2018-04-18:
I can't replicate this on MicroPython. I tried the Unix build and also on a Pyboard V1.1. Access to the latter was by rshell and also Miniterm.
```
MicroPython v1.9.3-547-g3d5d76f on 2018-04-11; PYBv1.1 with STM32F405RG
```
I also tried the Pyboard on Mu without issue (albeit my Mu is an old version).

So my guess is it's a CircuitPython issue and should be raised with Adafruit.

dpgeorge, 2018-04-20:
I would guess that this is an issue with Mu's REPL input window.  It may not capture/process the delete and backspace keys correctly (I remember early on in Mu's development that it didn't).  It can be tricky to get this kep mapping right and may depend on the OS that you run Mu under.

But uPy itself works ok so I'll close this issue.

closed 2018-04-20 by dpgeorge

dhalbert, 2018-04-20:
This is reproducible without Mu at all. Using Tera Term on Windows, DELETE does not delete forward. With Putty, it does. Other terminal programs on Linux and MacOS (screen, picocom) work fine. So I think it's some kind of key-mapping issue.

Sembazuru, 2018-04-20:
FYI, this has been raised to bug status in the CircutPython repository (see the above reference).
(Though, I see dhalbert should already know this as he has been co-assigned this issue over there.) 

Hopefully the CircuitPython folks well squash this bug and the fix will propagate upstream to micropython if applicable. (So would that actually be converting the bug into a spawning salmon?) ;-)

robert-hh, 2018-04-20:
It is definitely a key mapping issue of the terminal emulator. The backspace key is traditional Ctrl-H (0x08), but nowadays it's is more likely that the Backspace key sends DEL (0x7f), because Ctrl-H is used for Search/Replace in the PC GUI's. In contrast, the DEL key on ol terminals used to send DEL (0x7f), but PC keyboards send ESC[3~, and most terminal emulators do so. Micropython is set to that.
Some of the terminal emulators are configurable, some not.

robert-hh, 2018-04-20:
Just tried it: you can configure TeraTerm to send DEL with the Backspace key. The it will send the proper sequence with the DEL key.