micropython/micropython #3732: How to use DEBUG_VERBOSE to build micropython?

eminarcissus, 2018-04-20:
Hi,

I've written a C based python module for esp32, and after I have it compiled in the firmware, it said cannot find module. I did check there's module.o in application.map, and I've checked builtinimport there's a debug arguments named MICROPY_DEBUG_VERBOSE in the code. So I added -DMICROPY_DEBUG_VERBOSE in the makefile, but it failed to compile with error
malloc.c:(.text.m_malloc_maybe+0x18): undefined reference to `DEBUG_printf'
build/py/malloc.o:malloc.c:(.text.m_malloc_with_finaliser+0x1f): more undefined references to `DEBUG_printf' follow
Makefile:844: recipe for target 'build/application.elf' failed

So I'm wondering how to use this DEBUG VERBOSE flag?

eminarcissus, 2018-04-20:
> lib/util/printf.c has a definition of DEBUG_printf if
> MICROPY_DEBUG_PRINTERS is 1, but there are probably other ways, I assume
> -DDEBUG_printf=printf or so might also do the trick, not tested though

Weird, didn’t noticed this flag, thanks very much for the help :)
-- 
Narcissus Emi

:

dpgeorge, 2018-04-20:
The esp32 port doesn't use the internal (uPy-provided) printf code because it has its own printf that it needs to use.  You might need to add your own implementation of DEBUG_printf, just copy that in lib/utils/printf.c

Note that you can do `help('modules')` at the REPL to list known built-in modules.  And you must put your custom C modules in mpconfigport.h, in MICROPY_PORT_BUILTIN_MODULES

eminarcissus, 2018-04-23:
Got it working right now, that really helps, thx very much for the help :)

closed 2018-04-23 by eminarcissus