micropython/micropython #3739: ESP8266: OSError: [Errno 98] EADDRINUSE when rerunning code.

yeyeto2788, 2018-04-24:
Hello guys, 

I have been doing some scripts for my ESP8266 based board (nodemcu v2) and after successfully running the code for the first time, then upload changes (via Adafruit ampy tool) on the code and error is raised on the part of the code from the [**`MicroBoard/releases/MicroBoard.py`**](https://github.com/yeyeto2788/MicroPythonScripts/blob/master/MicroBoard/Release/MicroBoard.py) file in the `s.bind(addr)` method call.

MicroPython version installed on the board:
```
>>> import os
>>> os.uname()
(sysname='esp8266', nodename='esp8266', release='2.0.0(5a875ba)', version='v1.9.3-8-g63826ac5c on 2017-11-01', machine='ESP module with ESP8266')
```

Error raised on the REPL
```
>>> import MicroBoard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "MicroBoard.py", line 109, in <module>
  File "MicroBoard.py", line 88, in main
OSError: [Errno 98] EADDRINUSE
```

**Steps to reproduce:**
1- Upload the code
2- On the REPL do a `import MicroBoard`
3- WITHOUT unplugging the board, make some changes on the code (like one of the print statement)
4- Open the REPL and do again `import MicroBoard`

**Current behavior:**
Error is raised :(

**Expected behavior:**
No error should be raised.

Might it be due to not resetting the AP when the script starts and it is trying to reconnect to the same address?.

**NOTE:**
* This happens every time I try to run the code for the second time without unplugging the board.
* If I unplug the board wait 2 secs and then run the code it does not happen (This will be like the "first" time.)
* `ctrl + d` does not solve the problem (or at least the REPL is not usable in my case)

robert-hh, 2018-04-24:
Two comments:
- If you want re-import a module, you have to perorm a soft-reset in between. Ampy does that for you. But if you do it differently, don't forget it.
- the LAN stack remembers open sockets and open interfaces even through a soft reset. To bring that back to initial state, you have to perform a hard reset, either by taking the power of and on again, by pushing the reset button or pulling the reset line low. This is not comfortable, but unlikely to be changed.

yeyeto2788, 2018-05-02:
Hello guys, 

Another thing happened today, after executing the code and doing a `machine.reset()` the REPL would be unusable. 

```
>>> import MicroBoard
client connected from ('192.168.4.2', 42638)
client connected from ('192.168.4.2', 42639)
client connected from ('192.168.4.2', 42641)
client connected from ('192.168.4.2', 42643)
client connected from ('192.168.4.2', 42644)
client connected from ('192.168.4.2', 42645)
client connected from ('192.168.4.2', 42648)
client connected from ('192.168.4.2', 42649)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "MicroBoard.py", line 121, in <module>
  File "MicroBoard.py", line 104, in main
KeyboardInterrupt:
>>>
>>> import machine
>>> machine.reset()
>>>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31100, room 16
tail 12
chksum 0xe3
ho 0 tail 12 room 4
load 0x3ffe8000, len 1084, room 12
tail 0
chksum 0xc0
load 0x3ffe8440, len 3248, room 8
tail 8
chksum 0xe1
csum 0xe1
d▒▒|▒▒;snc▒▒
d▒
b▒dc䌜▒▒▒d
c▒lrll
d▒▒|▒▒snc▒▒ld▒p▒#▒c▒Ĝ▒▒c
c▒▒l;l▒l▒▒|▒▒s;gb▒▒$
▒▒
▒c▒
b쌜▒▒
l#d▒▒▒n▒{▒▒o|▒l$ld`▒▒r▒$▒d▒
$`▒▒s▒l▒l
l ▒▒{▒dǟ▒
ld`;l▒▒{d▒▒▒▒c
▒▒c▒c
br▒▒{
▒▒o▒▒'▒l▒▒l▒
$▒▒ll▒▒▒Č▒$▒'▒▒▒▒cl$▒{p
▒▒▒▒c
d▒b{$sdr▒g▒▒▒▒▒oܟ▒cr
▒▒c쎜Ä▒▒▒▒▒▒$lbd▒▒lg▒8{l▒lܟ▒#4 ets_task(40100130, 3, 3fff837c, 4)
```

Then I have load the code again and test the code. After 11 times sending information through the web page I got a different error.

```
>>> import MicroBoard
client connected from ('192.168.4.2', 42657)
client connected from ('192.168.4.2', 42658)
client connected from ('192.168.4.2', 42659)
client connected from ('192.168.4.2', 42660)
client connected from ('192.168.4.2', 42661)
client connected from ('192.168.4.2', 42662)
client connected from ('192.168.4.2', 42664)
client connected from ('192.168.4.2', 42665)
client connected from ('192.168.4.2', 42666)
client connected from ('192.168.4.2', 42667)
client connected from ('192.168.4.2', 42669)
client connected from ('192.168.4.2', 42670)
client connected from ('192.168.4.2', 42672)
client connected from ('192.168.4.2', 42673)
client connected from ('192.168.4.2', 42674)
client connected from ('192.168.4.2', 42675)
client connected from ('192.168.4.2', 42676)
client connected from ('192.168.4.2', 42677)
client connected from ('192.168.4.2', 42680)
client connected from ('192.168.4.2', 42681)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "MicroBoard.py", line 121, in <module>
  File "MicroBoard.py", line 118, in main
OSError: [Errno 104] ECONNRESET
```

I'm not quite sure if this issue is highly related to the issue #2055 or even related to #2661

Regarding those issues, seems like both of them have something in common which is making/sending requests in a short period of time and the issue #2661 does states that closing the socket after sending the information. As far as I can see I'm doing so but I also got the error.

So, to summarize:
* The initial error (`[Errno 98] EADDRINUSE`) seems to be fixed by just doing a `machine.reset()` or pulling the reset line low as @robert-hh mentioned before.
* After doing the hard-reset the REPL would break, giving you NO possibilities to execute code and so on.
* After loading the file again (via ampy which does a soft-reset), executing the code and sending information through the web page I'll get a (`[Errno 104] ECONNRESET`) error.

I hope I'm not confusing anybody with all this information and trying to link previous issues on the repo.

Best regards.

yeyeto2788, 2018-05-03:
Hello @dpgeorge, thanks for your reply.

**1.** I'll give it a try to `socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)` before the `socket.bind(addr)` to see if the error `[Errno 98] EADDRINUSE` is not reproduced anymore.

**2.** What I meant by "unusable" was that it will give me a gibberish output after the `machine.reset()` the output is like the one and pasting below and the REPL does not accept any keyboard event on the terminal. I tried soft-reset (`ctrl+d`) or other shorcuts availables to use in the REPL (`ctrl+b`, `ctrl+e`, `ctrl+c`) and none of those seemed to do nothing on the REPL.

   If you take a closer look at the output, you can notice that the `>>>` are not even present after the last line coming from the board, so that also made think something went wrong.

```
>>> import machine
>>> machine.reset()
>>>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31100, room 16
tail 12
chksum 0xe3
ho 0 tail 12 room 4
load 0x3ffe8000, len 1084, room 12
tail 0
chksum 0xc0
load 0x3ffe8440, len 3248, room 8
tail 8
chksum 0xe1
csum 0xe1
d▒▒|▒▒;snc▒▒
d▒
b▒dc䌜▒▒▒d
c▒lrll
d▒▒|▒▒snc▒▒ld▒p▒#▒c▒Ĝ▒▒c
c▒▒l;l▒l▒▒|▒▒s;gb▒▒$
▒▒
▒c▒
b쌜▒▒
l#d▒▒▒n▒{▒▒o|▒l$ld`▒▒r▒$▒d▒
$`▒▒s▒l▒l
l ▒▒{▒dǟ▒
ld`;l▒▒{d▒▒▒▒c
▒▒c▒c
br▒▒{
▒▒o▒▒'▒l▒▒l▒
$▒▒ll▒▒▒Č▒$▒'▒▒▒▒cl$▒{p
▒▒▒▒c
d▒b{$sdr▒g▒▒▒▒▒oܟ▒cr
▒▒c쎜Ä▒▒▒▒▒▒$lbd▒▒lg▒8{l▒lܟ▒#4 ets_task(40100130, 3, 3fff837c, 4)
```

   **NOTE:** Powering off and then power on the board by just plugging it out and in will do the trick to have the REPL working again. If you want me to try something else or want me to create another issue for this just let me know.

**3.** Thanks for that comment, I always thought that as the connection was strong enough there shouldn't be any problem and reading the other issues made think that maybe it was caused by the time between requests.

Regards.

dpgeorge, 2018-05-04:
> What I meant by "unusable" was that it will give me a gibberish output after the machine.reset() the output is like the one and pasting below and the REPL does not accept any keyboard event on the [...]

The gibberish is normal, it's the ESP boot code printing stuff out at a different baudrate than 115200.  It could be that your terminal gets confused by this gibberish and breaks.  I suggest trying a different terminal/serial program.

dpgeorge, 2020-06-09:
Closing due to inactivity.

closed 2020-06-09 by dpgeorge