micropython/micropython #3740: Interest in PPPoS?

MrSurly, 2018-04-24:
I have an upcoming project where I'm using PPP and a cellular module for network connectivity.  Was wondering:

* If there's interest in using PPPoS
* If a patch would be accepted for PPPoS

The idea being the user code does all the setup work for the modem, and hands off a UART object to a `network.PPP` object.

Thoughts?

dhylands, 2018-04-24:
I'd find this to be interesting. This would give network access to pyboards connected to say a Raspberry Pi.

MrSurly, 2018-04-24:
> This would give network access to pyboards connected to say a Raspberry Pi.

Right, sorry.  Forgot to mention I had the ESP32 in mind.  Having said that, I suspect it could still be done, since PPP is part of LWIP.

peterhinch, 2018-04-25:
Hasn't this been done [here on the Loboris port](https://github.com/loboris/ESP32-PPPOS-EXAMPLE)?

pmp-p, 2018-04-25:
it was relevant for me but in a hurry flying over the stack of issues i may have probably wrong read instead :
Hasn't this been done here on the Loboris port *like everything else* ?

MrSurly, 2018-04-25:
* I have no interest in the Loboris derivative
* The link went to a plain C ESP-IDF example of PPPoS

pmp-p, 2018-04-25:
afaik micropython can be built as an esp-idf component, so why lose time with not-invented-here ?

dpgeorge, 2018-04-26:
Yes that would be interesting to have.  It would be great to do it in a general way (eg building upon extmod/modlwip.c) but I guess that's not possible because the ESP32 uses it's own custom lwIP?

dpgeorge, 2018-05-01:
That looks pretty reasonable.

Did you see that there is (basic but working) SLIP support in extmod/modlwip.c?  That's similar to PPP but the code there is very specific to lwIP.  And the network module is arguably the right place to put these kind of objects because they are configuring a data link layer.

dpgeorge, 2018-05-18:
> Seems the LWIP that comes with MP is pretty old (2012) compared to the LWIP that comes with the ESP-IDF (2016). What are the ramifications of using a newer LWIP?

I actually just updated it yesterday to the latest stable lwIP 2.0.3.

dpgeorge, 2018-10-19:
network.PPP was implemented in 3c6f639aa58d0558a744af2f2fc32a6debf55c81

closed 2018-10-19 by dpgeorge